MoonshineSG/OctoPrint-Mobile #1: Loading ...

arnaud-42, 2016-09-14:
I have some problems with your app.

My setup : 
Octoprint on a raspberry behind a router.
My router forward every connexion to https to my octoprint.
Https certs are letsencrypt certs ... so no self signed certs.
octoprint has a login/password (http auth basic)

When i try to run the app ... I have a loading text ... and that it.

Any idee how to debug ?

MoonshineSG, 2016-09-14:
I assume you setup the API key in the 'settings'?
Can you access the web OctoPrint using a browser on your mobile on https? No cert error?

arnaud-42, 2016-09-14:
Yes API key is ok.
I can access my octoprint from any browser without any error.

MoonshineSG, 2016-09-14:
I see... don't laugh, but try to shake your device... it will reload the page... 

Btw, what device and OS version. are you using?

arnaud-42, 2016-09-14:
Ios 10 iphone 6s plus.

MoonshineSG, 2016-09-14:
Can you paste the logs from OctoPrint? 

If shake didn't work, can try to kill the app and restart it...

MoonshineSG, 2016-09-14:
Would this apply in your case? 
https://community.letsencrypt.org/t/ios-native-apps/7827

arnaud-42, 2016-09-14:
My cert is ok (concatenation oc cert and fullchain.pem) ...
I will look at my log to night ... maybe i can see something.

robinJKU, 2016-09-16:
Same problem here! The funny thing is I removed the plugin from octoprint, restarted, than the app loaded but without icons.
Now I reinstalled the plugin and the app hangs on the Loading... screen.
IOS 10 on Iphone 7

MoonshineSG, 2016-09-16:
The app will do some caching but if you really removed the plugin and restarted OctoPrint, the app should have complained that the plugin is missing... I assume you're both on version 1.3.1 ?

MoonshineSG, 2016-09-16:
And really??? iPhone 7 so fast? Obviously not yet tested :) but iOS 10 should be ok...

luke321, 2016-09-16:
Had to remove the other comment, was logged in with my work account...
I am using the current octoprint devel branch and I used the 1.3.3 Version of your plugin.
It actually complains about the missing plugin but shows the correct temperatures first.

And yeah arrived today, very nice phone :)

luke321, 2016-09-16:
Found 2 thinks in my logfile:

`2016-09-16 16:36:54,539 - octoprint.server - WARNING - requests/urllib3 will run in an insecure SSL environment. You might see corresponding warnings logged later ("InsecurePlatformWarning"). It is recommended to either update to a Python version >= 2.7.9 or alternatively install PyOpenSSL plus its dependencies. For details see https://urllib3.readthedocs.org/en/latest/security.html#openssl-pyopenssl`

and

```
2016-09-16 16:38:04,502 - octoprint.server.util.sockjs - INFO - New connection from client: 10.0.0.6
2016-09-16 16:38:13,194 - octoprint.plugins.nautilus - INFO - X-Forwarded-For : [['10.0.0.6']]
2016-09-16 16:38:19,803 - octoprint.plugins.nautilus - INFO - X-Forwarded-For : [['10.0.0.6']]
2016-09-16 16:38:20,021 - octoprint.plugins.nautilus - INFO - X-Forwarded-For : [['10.0.0.6']]
2016-09-16 16:38:21,577 - octoprint.plugins.nautilus - INFO - X-Forwarded-For : [['10.0.0.6']]
2016-09-16 16:38:23,373 - octoprint.plugins.nautilus - INFO - X-Forwarded-For : [['10.0.0.6']]
```

My Raspberry pi is running without a monitor, so there should be no X-Server Running, so X-Forward shoul not be working? Or am I misinterpreting this?

Whole log:
[octoprint (2).zip](https://github.com/MoonshineSG/OctoPrint-Mobile/files/477307/octoprint.2.zip)

MoonshineSG, 2016-09-17:
No, the X in this case does not refer to an X server, but to a proxy header. But your logs helped to identify the problem. I was only checking for 192.168 addresses, not the 10.

Should be ok now.

MoonshineSG, 2016-09-17:
@arnaud-42, see if this helps your problem as well (fingers crossed)

luke321, 2016-09-17:
It is working now! Thanks for the quick fix.

MoonshineSG, 2016-09-17:
@luke321 glad to hear it works and thanks for finding an error. :D

closed 2016-10-13 by MoonshineSG