MoonshineSG/OctoPrint-Mobile #2: U need to Install Nautilus

Peacens, 2016-10-03:
Hi There, Thanks 4 Time

I tried to Install nautilus, get under addons the Url with the Master.zip, but on Install it won,t
Have u How To to Install it on Octoprint?

best regards

MoonshineSG, 2016-10-09:
what version of OctoPrint are you running  ?

eric777, 2016-10-17:
Tried installing Zip file in plugins and got this message 

"Something went wrong Installing the plugin from URL "unknown" failed: Could not parse output from pip"

and the app says "This application requires the plugin "Nautilus" to be installed on OctoPrint."

OcotPrint says "everything is up to date"
OctoPrint: 1.2.16 

What should I try to do next?

MoonshineSG, 2016-10-18:
as mentioned in the plugin description it requires a "devel" branch version

angelocarlotto, 2016-10-18:
I am facing same problem.

My octoprint version also is  1.2.16

Cant you make the plugin available also to this version?

I feeling a lit bit strange, because i already paid the iOS app. And would like to use octopring stable version. Not dev.

Thanks

MoonshineSG, 2016-10-18:
Unfortunatly I can't. The stable version doesn't support custom UI plugins. Once that feature is promoted from devel to release, I will make sure Nautilus works. 

In the description of the iOS app I did mentioned that it requires the devel branch. Hopefully soon it can be used on the stable release version as well.

angelocarlotto, 2016-10-18:
All right.

You are right, soon it will became release version.

Keep improving man, I bet it's gonna be a very good solution to remote 3D printing and monitor.

Cheers

angelocarlotto, 2016-10-18:
Man, how can i get the imagem from deve just like i get from the website?

is that possible?

Can you share the link to download?

Thanks

angelocarlotto, 2016-10-18:
Can i use one of this: http://docstech.net/OctoPiMirror/nightly/#rpi-nightly ?

MoonshineSG, 2016-10-19:
@angelocarlotto, firstly, the link above is not up to date (the last 'nightly is from June). The most updated is http://gnethomelinux.com/OctoPi/nightly/
 
But, even these use the main branch, meaning they are not suitable for Nautilus. 

You might want to read this: https://github.com/foosel/OctoPrint/wiki/FAQ#how-can-i-switch-the-branch-of-the-octoprint-installation-on-my-octopi-image

MoonshineSG, 2016-10-20:
Looks like it's coming sooner than expected :D 

https://github.com/foosel/OctoPrint/releases/tag/1.3.0rc1

I'll have to run some tests but I don't foresee  many problems...

angelocarlotto, 2016-11-03:
Man, do you know what would be awesome? You could make available a ready to download image of this version of the octoprint. This version of octorpint that is compatible with your app.

I am pretty sure your costumer would love it even more Nautilus.

Beginning from me.

Cheers.

MoonshineSG, 2016-11-03:
Ya.... That's not gona happen. :D 

That would actually mean that I maintain a separate OctoPrint release cycle... Imagine the amount of additional time I'd spend on this.  And there are really very very little benefits from it.  

OctoPrint is very well documented and the installation/upgrade procedures relatively simple. There is even that ready made RPi image... 

Now that the Nautilus is in the official repository, with a few clicks you have `Nautilus` up and running ( just update the `settings.ini` with your custom gcodes! ) and after that it's on auto pilot :D

angelocarlotto, 2016-11-03:
Definitely you are a great developer.

Peacens, 2016-11-03:
1.2.16
I use the octoprint Image from foosel.
I changed the settings to develbranch, doesn`t work. 
What's next?

MoonshineSG, 2016-11-03:
@Peacens,  as stated, it requires 1.3.0rc1 or above or devel branch

closed 2017-01-09 by MoonshineSG