MoonshineSG/OctoPrint-Mobile #4: Invalid API Key?

1n5aN1aC, 2016-11-09:
This looks very promising, but the app is now giving me a "Invalid API Key" Message and not working, even after clearing the key, and re-copying it from the browser.

Interestingly, it seemed to work at first, but upon a reboot of the octoprint-running rasPi, I have not been able to get it to work any longer...

Any ideas what could be causing this?

MoonshineSG, 2016-11-10:
I am using the same branch, no problem. I don't have a user setup, but that shouldn't be an reason.

You're using the key under "Settings/API Key" and the "Enable" is checked. Right? 

A look at the log files might help.

AlexanderPro29, 2016-11-16:
I also have this problem. The application has stopped working. Tried to change the API key - to solve the problem failed.

1n5aN1aC, 2016-11-16:
In my case, the API key got in a strange quantum state where it displayed, yet none would be saved to the config file.

I haven't had time to fix it quite yet, just providing more information that might help you.

MoonshineSG, 2016-11-17:
@1n5aN1aC , thanks. You can try to edit the config.yaml manually and add in the api key. 

```
api:
  key: XXXXXXXXXXXXXXX
```

@AlexanderPro29, similar,  check the config.yaml and see if the key is there. if not, add it in manually.

AlexanderPro29, 2016-11-17:
@MoonshineSG i'm checked the config.yaml (./home/pi/.octoprint/config.yaml) and see api key there. It coincides with the key in the iphone settings, but the app says "invalid API key".

AlexanderPro29, 2016-11-17:
@MoonshineSG Is the key length is important in symbols?

MoonshineSG, 2016-11-17:
The api key is handled by OctoPrint...  
Can you check if there are any spaces before or after the key (in the iOS app settings) ?

AlexanderPro29, 2016-11-17:
@MoonshineSG I checked and found no spaces before / after the key or inside.

MoonshineSG, 2016-11-17:
[image]

Just to be sure... This is the key you added. And it's enabled.

MoonshineSG, 2016-11-17:
what version of OctoPrint ?

AlexanderPro29, 2016-11-17:
Version: 1.3.0rc1.post0.dev0+gb38d181 (rc/devel branch)

AlexanderPro29, 2016-11-17:
[image]
I'm open settings and see them

MoonshineSG, 2016-11-17:
you have a "n/a" instead of a key ...

AlexanderPro29, 2016-11-17:
I filled out the field API key in the same code as in the iphone settings and hit "save", but the problem has not been solved.
If it can help, I have a public IP address, and API key of my OctoPi.

MoonshineSG, 2016-11-17:
sure, i can try

AlexanderPro29, 2016-11-17:
address pro29.telenet.ru:5000 
I created a user name/pass test/test
API key 7207F81507C346559C15794A5DFD1C98

AlexanderPro29, 2016-11-17:
After restarting the server I have just successfully connected! The problem is solved, but it is not clear what it was. Server I rebooted before.

MoonshineSG, 2016-11-17:
yes, i can see your printer :D 

now change your api key quickly!! :D

AlexanderPro29, 2016-11-17:
Of course, I'll change the code)

MoonshineSG, 2016-11-17:
anyway, I'm glad you got it sorted out.

AlexanderPro29, 2016-11-17:
After replacing the API key problem persists. The key is stored in ./.octoprint/config.yaml, but for some reason that does not work ... I have removed completely the key in the web interface settings, copy it to the clipboard and through the web interface System> restart OctoPrint. Then again went into the settings and recorded the same key from the clipboard, and then to re-System> restart OctoPrint. Then again earned.

MoonshineSG, 2016-11-17:
seams to be a problem with the apikey  in OctoPrint. I'll check with the OctoPrint developer..

AlexanderPro29, 2016-11-17:
I am now convinced that adding / removing user helps in solving the problem. That is to API key to work, I added a new user via a web interface and rebooted OctoPrint, then the user can be deleted.

MoonshineSG, 2016-11-17:
the api key should be user independent.. I don't even have a user in my octoprint... no login for me... :D

AlexanderPro29, 2016-11-17:
However, this affects access. Even after a successful login and attempts to reconnect after a while, again, there is an error Invalid API key. Then I do OctoPi restarts and the plugin has successfully authenticated.

MoonshineSG, 2016-11-17:
That is not a normal behaviour... I have never had this problem. 
I'm off on a short holiday but I'all have look at the problem when I'm back next week.

1n5aN1aC, 2016-11-24:
Yeah, I'm almost convinced this might be an issue in Octoprint itself perhaps.

I'm still having issues, have tried restarting the octoprint computer, re-entered API key, API is enabled, key appear in web GUI, key appears in config.yaml, same error.

I then changed the API key in the web GUI, clicked save, the config.yaml was updated, killed the app on my phone, changed the API key setting on my phone, and opened the app back up.  same problem.

I am honestly very confused as to why it worked int he very beginning, but then hasn't worked since.

MoonshineSG, 2016-11-24:
I know this is not the most fun thing to do, but you might wana try to do a fresh OctoPrint setup.

AlexanderPro29, 2016-11-24:
I also still have problems with a key. The application can successfully join, but after a while I receive an error when reconnect.

1n5aN1aC, 2016-11-24:
I'll try a reinstall once I get some more SD cards that I've been planning on buying anyways.  :P

MoonshineSG, 2016-11-24:
@AlexanderPro29 , "an error" ? can you give the exact message and, if possible, under which exact conditions? What version of OctoPrint ? 
Maybe raise this under the OctoPrint support ? Seams to be something on their side. (I'll check as well, just in case)

@1n5aN1aC,  please let me know how it goes. Hopefully it helps.

AlexanderPro29, 2016-11-24:
@MoonshineSG I meant the message "Invalid API key". Version: 1.3.0rc1.post0.dev0+gb38d181 (rc/devel branch)
I have yet to start to understand how it works github and English is not my native language. I apologize if my text is difficult to understand.

MoonshineSG, 2016-11-24:
@AlexanderPro29 , no worries, Just needed more info to try to figure it out.  I'll check with OctoPrint to see if there's something they can help with.

AlexanderPro29, 2016-11-25:
Today I installed Version: 1.3.0rc2 (rc / devel branch) to the latest version Raspbian, API key works. Later I will try to update the version installed on OctoPrint OctoPi.

1n5aN1aC, 2016-11-26:
@AlexanderPro29 can you confirm that the app continues to work after you did the reinstall?

AlexanderPro29, 2016-11-27:
@1n5aN1aC Yes, the app works after a complete reinstall. (Formatting a SD card and a new installation)

AlexanderPro29, 2016-11-28:
Today the problem is back. And Sli3cr program successfully see the printer, and Nautilus says "Invalid API key". I did not make any changes to the program, I do not understand what is connected with this error.

MoonshineSG, 2016-11-29:
And the settings still shows the same API key? Did OctoPrint somehow generated a new API key?

Monox83, 2016-12-07:
Hi, I had the same problem.
My Raspberry Pi is equipped with WLAN with Static IP. In my router it was with a second IP, this is not displayed in Pi, with this Andresse it works without problems.

AlexanderPro29, 2016-12-08:
The problem in a mobile application, because other programs have successfully connected using the API. In the settings OctoPrint API has not changed. Sli3cr connects successfully, at the same time Nautilus reported "Invalid API key". The recently published update to OctoPrint rc03, the problem persists and the Nautilus is not working.

MoonshineSG, 2016-12-08:
If using the same key and the same URL, it should work fine.  The message "Invalid API key" is from OctoPrint.. 
Can you share again your url/apikey ?  Let me try so see if I can spot anything

AlexanderPro29, 2016-12-10:
I apologize for the delay in replying. Again I set up access to the address pro29.telenet.ru:5000 API key 55468C2C5CAA4C4F8A49EA71593943B9
I also created a test / test user
The printer is currently not connected due to a problem with the printer arduino. But the camera is connected.

MoonshineSG, 2016-12-10:
i can see the live feed....

analogmonster, 2016-12-10:
I am another user with the same issue.  I am on master 1.3.0, and nautilus 1.5.4 plugin on a PI3.  IOS app is up to date.  I did a clean octopi install yesterday (had a problem gonig from dev to master so did a clean master).  I dont have any other issues.  I have used the Octo app for IOS which also uses the API key, which I have to keep using until nautilus is behaving properly.  Octo is fully functioning.  However if I go to nautilus I get invalid API key.

I have tried changing the API key manually.  Once I did this Octo asked me to reconfigure, I gave it my new custom key, copied it to clipboard.  Tested and octo worked.  Do a paste of the key (which I am now certain is valid thanks to octo) into nautilus settings and still get invalid API key.  No spaces before or after.

It seems most likely this is at the IOS end rather than an Octoprint API key issue since I have no issues with another app.

MoonshineSG, 2016-12-10:
@analogmonster And you're trying to access it via the same wifi ? 

@analogmonster @AlexanderPro29 can you paste your config.yaml  ?

MoonshineSG, 2016-12-10:
@analogmonster what ios device  ?

analogmonster, 2016-12-10:
same wifi same device same location.  I'm on octopi.local at the moment.  Also I always properly close the app before changing the API. Iphonr 6s plus  on 10.1.1

analogmonster, 2016-12-10:
and I did have it working on my previous install when I was still on dev branch.  How do I get to my config.yaml, do you want me to ssh in or is there something I can do within octoprint?

MoonshineSG, 2016-12-10:
Omul via ssh

analogmonster, 2016-12-10:
it will have to wait a while I'm afraid as I am mid print now but will send it ASAP

MoonshineSG, 2016-12-10:
It's midnight here so... take your time. I'll read in the morning :)

AlexanderPro29, 2016-12-10:
@MoonshineSG I open the link http://pro29.telenet.ru:5000 in a browser and see UI. Nautilus does not see anything except for the video stream, if connected externally. I installed all strictly according to instructions.
My config.yaml file (renamed to .txt):
[config.yaml.txt](https://github.com/MoonshineSG/OctoPrint-Mobile/files/643938/config.yaml.txt)
My haproxy config file (renamed to .txt):
[haproxy.cfg.txt](https://github.com/MoonshineSG/OctoPrint-Mobile/files/643942/haproxy.cfg.txt)
My iOS device is iphone 5s
@analogmonster I use Bitvise SSH Client software for connecting to the raspberry and it is very convenient to copy the necessary files via SFTP.

analogmonster, 2016-12-10:
@AlexanderPro29 bitvise looks cool, read the message too late for this time but I will download a trial later.  I can see your octoprint UI at that address too so its definitely working.

@MoonshineSG see below.  obviously the xxxx ones I have changed myself

accessControl:
  salt: xxxx
api:
  allowCrossOrigin: true
  key: 'xxxx'
plugins:
  announcements:
    channels:
      _important:
        read_until: 1478614200
      _plugins:
        read_until: 1481241600
  cura:
    cura_engine: /usr/local/bin/cura_engine
  discovery:
    publicPort: 80
    upnpUuid: xxxx
  nautilus:
    _settings_version: 3
  pushover:
    events:
      PrintDone:
        priority: '0'
      PrintFailed:
        priority: '0'
    sound: spacealarm
    token: 'xxxx'
    user_key: xxxx
  softwareupdate:  
    _config_version: 4
    check_providers:
      displayprogress: displayprogress
      navbartemp: navbartemp
      pushover: pushover
      slicer: slicer
      title_status: title_status
    checks:
      octoprint:
        update_folder: /home/pi/OctoPrint
printerProfiles:
  default: anycubic
serial:
  baudrate: 230400
  port: /dev/ttyACM0
server:
  commands:
    serverRestartCommand: sudo service octoprint restart
    systemRestartCommand: sudo shutdown -r now
    systemShutdownCommand: sudo shutdown -h now
  firstRun: false
  secretKey: xxxx
  seenWizards:
    cura: null
slicing:
  defaultProfiles:
    cura: delta_pla
system:
  actions: []
webcam:
  ffmpeg: /usr/bin/avconv
  snapshot: http://127.0.0.1:8080/?action=snapshot
  stream: /webcam/?action=stream

MoonshineSG, 2016-12-11:
@AlexanderPro29 : i can see the octoprint normal UI , and the Nautilus. I dont get API error... 

On your iphone you use the external url (http://pro29.telenet.ru:5000/) or you use the internal network IP ?

AlexanderPro29, 2016-12-11:
I used the internal static IP address 192.168.1.153 to connect. Now I tried it and found that an external location is really connected and I see a picture with the camera. But with the internal address is an error "Invalid API key". I also tried other device iPad 2 and there is the same problem.
For outside access to my router set up port forwarding on port 5000 to port 80 internal address. I also tried the settings Nautilus forcibly put port 80, but nothing has changed.

AlexanderPro29, 2016-12-11:
In Sli3cr program exhibited the same settings to the internal IP address, and it successfully connects and sends the files to OctoPrint.

MoonshineSG, 2016-12-11:
I am able to connect and see the Nautilus interface. I really don't understand why you have this problem...

AlexanderPro29, 2016-12-11:
My guess is that the authors OctoPrint something changed and now access to printer management is performed once on the other? I noticed that in the user settings the opportunity to destination user API key.

MoonshineSG, 2016-12-11:
I'll ask, but unlikely.
So you use the same API key for both, slicer and Nautilus?

AlexanderPro29, 2016-12-11:
Yes, I use the same key, the same internal address. I have already checked many times, and tried to change the API.

[44 more replies]

closed 2017-01-24 by 1n5aN1aC