MoonshineSG/OctoPrint-Mobile #5: Documentation issue

Pipshag, 2016-11-16:
Hi!

I'm having trouble locating or understanding how I would set the variable for @beep. I don't want to send a standard M300 as a beep but instead I'd like to use M300 P10 - and also have the possibility to disable the beep completely. 

How is this achieved? Tried setting beep : 10, beep : P10 and @beep : P10 in the settings file without any luck.

MoonshineSG, 2016-11-17:
in order to have the octoprint play the sounds using "M300 @beep" you need another plugin (https://github.com/MoonshineSG/OctoPrint-Sound). The sound plugin intercepts the M300 comands and play the coresponding message and does not send the command to the printer. 

Hope this helps.

Pipshag, 2016-11-17:
Thanks for replying.

I have a buzzer, which is quite common, and thought that it could be a welcomed feature, to be able to change the M300 duration or override the command altogether. It also beeps on temperature changes and wakes up the whole family.

Running a M300 gives a long, horrible, beep from my machine. While running M300 P10 gives a quick and short confirmation tone. 

Removed all references to M300 in the source code so I'm not bothered at the moment.

MoonshineSG, 2016-11-17:
Runing Marlin ? I think they made some changes recently on the "buzzer" support... 
I used to have that as well and as you say, it was horible. Thats why I added real speakers on my RPi and wrote the sound plugin... Now it plays any MP3 (i created some tiny tunes for the events I am interested in)

Pipshag, 2016-11-17:
Yes, running Marlin. Horrendous sound in the standard! But a confirmation beep is nice, P10 or a short duration was really nice to use.

As for the beeps, haven't looked through the code fully (just stripped the M300 @beep) but I'd think that scraping a value like @tone_temp and perhaps @tone_confirm from the settings to a variable would be OK. The settings is user-reachable and would give everyone a chance to customize it.

Perhaps even with a  comment (or updated documentation) to give a hint about your OctoPrint-Sound plugin?

MoonshineSG, 2016-11-17:
hmmm... which M300 are you refering to ? 

(documentation.. yes, a necesary pain in the ***...but... i'll add some notes as and when...)

Pipshag, 2016-11-17:
What I removed was the commands in models.js

'M300 @temperature_bed', 'M300 @temperature_extruder', 'M300 @beep'

And I'd suggest using a method to override the @temperature_bed. That would make it possible to pickup the sounds for the other plugin as well as using a buzzer with both ON/OFF and duration.

In settings.ini: 
; Sound settings
{{temperature_bed}} : @temperature_bed
{{beep}} : P10

etc

MoonshineSG, 2016-11-17:
Ah... those.... 
I still have a few hardcoded gcodes that havent found their way to the settings.ini.
Eventually there will be no gcode in the JS files...

MoonshineSG, 2016-11-17:
as all my plugins, this was writen for my own use and grew into smth else... so it goes through a slow process of extarcting customizable parts. slow and painful :D

Pipshag, 2016-11-17:
I have total understanding for that, I'm a web developer and whenever I do a project for my own sake it's always lacking documentation and have nice tie-ins that are a pain to break up. :D

So perhaps you can tag the issue as a future request and we'll leave it at that for the moment?

MoonshineSG, 2016-11-17:
beside the above mentioned issue, how is it so far ? dont really get fedbacks and I'm curious how others feel about it  (i use it everyday and loving it ... hahahaha)

Pipshag, 2016-11-17:
I thought you'd get a ton of feedback through the AppStore. It's a good companion!

It's really nice and I love it. 
- Quick and responsive. 
- Good instructions inside the app.
- Thought out "at a glance" page.
- Separate section just for Z-offset - LOVING IT. :D
- Camera section looks nice without clutter.
- Configurable Macros with ability to show them.

Drawbacks
- Can't queue commands for height adjustments.
- Temperature sliders should snap harder. Difficult to get a precise 200C.
- Documentation for the module is kind of lacking.

For the documentation part I think it would be good to put in a Requirements before Updates in the readme. 

Requires devel-branch of OctoPrint and a link on how to switch (l thought there was some documentation from OctoPrint on how to do this but I couldn't find any just now). Requires companion app on OctoPrint. Requires Switching  plugin for features X, Y, Z. Requires OctoPrint-Sound for playing MP3 on RPi instead of Beeps with buzzer.

For added usability perhaps a section with a list of key points or features above requirements instead of having them unordered.

Pipshag, 2016-11-17:
Unfortunately I have no experience in iOS-coding or I would fork the repo and help out. Not that there's much to do, really... it's a solid product.

MoonshineSG, 2016-11-17:
through appstore i got only 3 negative review (form people not willing to read instructuons and not understanding what it was all about... moving on ! :D )

Thanks for the feedback.  Nice tho know at least one more person enjoys it. The z-offset was the reason why this started. 

Need more details on : 
- queue comands for height adjustment? 
- sliders can be easlily made more "snappy" but then it would be impossible to get a ... say 198C... 
- oh.. documentation.... right. 
- revert the XYZ-axises ?

No, it's not available for the 1.3.0rc1 onwards and it's basically a webbased (90% JavaScript), not iOS app... The app is just to hadle connection, screen lock on landscape so more content can be displayed (one of the reviewers complained about that!) and some other minot things, but that is mostly stable and won't change unless there are bugs...

MoonshineSG, 2016-11-17:
you can open request/issues with detail description of wanted enhancements and i'll try to accomodate in the limit of my free time

Pipshag, 2016-11-17:
Too bad about the negative review, I'll leave a positive one soon!
Just a quick note about 1.3.0.RC1 - do I need to lock the version because it breaks on the latest commits?

I'll put up some new issues then!

MoonshineSG, 2016-11-17:
breaks on the latest commits ? oh... I'm using devel branch, but I have a test RPi on RC1.. I'll have a look.

Pipshag, 2016-11-17:
Ah sorry, I haven't tested. Just wanted to know you knew that it did and we needed to lock the version to 1.3.0.RC1.

Nevermind! :D

Pipshag, 2016-11-28:
This seems to be solved as a user workaround - simply use M300 with a user specified variable, for instance beep = P10.

closed 2016-11-28 by Pipshag

MoonshineSG, 2016-11-28:
except for the "hardcoded" buttons. but soon everything will be in the settings.ini ("soon" is a relative term)

MoonshineSG, 2016-12-01:
the command executed by the buttons have been moved to `settings.ini`
this will allow full customisation. enjoy!