MoonshineSG/OctoPrint-Mobile #8: Snap Temperature settings

Pipshag, 2016-11-17:
**Temperature settings should snap**
It's quite difficult to get the correct temperature to preheat quickly. The sliders should snap easier to the preset intervals.

It would also be good to have an array of values for the temperature sliders that we utilize and that's configurable from settings.ini. This would make it easy to set up your own temp sliders, including having one for cleaning the nozzle, doing cold pulls. Would also make it possible (while keeping down the risks for people without firmware safety) to preheat to temperatures required for PC-ABS from the app.

MoonshineSG, 2016-11-17:
"array of values for the temperature sliders" sound like a good aproach.

MoonshineSG, 2016-11-26:
see https://github.com/MoonshineSG/OctoPrint-Mobile/releases/tag/1.5

closed 2016-11-28 by Pipshag

MoonshineSG, 2016-11-28:
let me know how it works.