MoonshineSG/OctoPrint-Mobile #10: iOS app doesn't work

JosvdP, 2016-12-08:
I installed the latest iOS app (and paid money for it...) but it doesn't get past the startup screen? It says "Loading" very quickly and then it says "Before you can use this application, the Nautilus plugin needs to be installed on Octoprint and you need to setup the Octoprint server URL and the API KEY in the settings of this app".

iPhone 6, iOS 10.0.2

MoonshineSG, 2016-12-08:
Did you install and configure the plugin?

JosvdP, 2016-12-08:
Yes. I'm not sure though what you expect me to configure? There is a default settings text that I haven't changed.

MoonshineSG, 2016-12-08:
Did you install the plugin?

JosvdP, 2016-12-08:
Yes I installed the plugin, version 1.5.4

MoonshineSG, 2016-12-08:
Instead of me asking so many questions, why don't you tell me excatly what you did.

MoonshineSG, 2016-12-10:
did you managed to get this to work ?

JosvdP, 2016-12-12:
Sorry, I completely misunderstood the instructions. I found the settings in the iPhone main settings. I thought this was confusing as I was expecting to have to put the settings of the API key into the app when it started.

However, one hurdle I have come across is the fact that the app doesn't handle https. I have Haproxy configured but this prevents the app from working.

MoonshineSG, 2016-12-12:
If the app doesn't have the API key it would prompt you when it starts. "Shake to open settings" But it's good you eventually found it.

As for https,  I'm not sure why you say it's not handled? In the settings it states "http or https" URL... I am always using https. 
If you use a self signed cert, please read the wiki on GitHub related to this.

JosvdP, 2016-12-13:
I've seen the wiki yes. I use haproxy with a login but I haven't made a self signed certificate. It just asks for my username and password when I go to the IP address.

By the way, the app didn't say "Shake to open settings".

MoonshineSG, 2016-12-14:
The "shake to load settings app" should be the first message you see. What version of the application do you use? 

When you say "ask for username and password" is the OctoPrint credentials or http authorisation? Regardless which version you use, http authorisation is not supported.

closed 2017-01-10 by MoonshineSG