MoonshineSG/OctoPrint-Mobile #11: Plugin not loading

rberg11, 2016-12-09:
Hello! It appears with the new octoprint 1.3.0 the plugin doesn't want to load? Perhaps I'm doing something wrong. New install, I tried installing the plugin through the plugin manager and it said it downloaded but after a restart does not load. So uninstalled via pip and then tried to install it via pip outside of octoprint.

I am running octoprint on a windows box and the other plugins I have tried load fine after a restart of the server.

MoonshineSG, 2016-12-09:
I have just updated to 1.3 and the plugin loads just fine.  Did you have the plugin running on an older version  ? Do you see any errors in the log file ?

rberg11, 2016-12-09:
Yeah fresh 1.3.0 install. Do you know where I can find the log to paste here?

MoonshineSG, 2016-12-09:
either via SSH or in the OctoPrint settings

rberg11, 2016-12-09:
[octoprint.txt](https://github.com/MoonshineSG/OctoPrint-Mobile/files/641097/octoprint.txt)

MoonshineSG, 2016-12-09:
ok. I see why it doesn't load. 
```
2016-12-09 20:41:43,428 - octoprint.plugin.core - ERROR - Error loading plugin nautilus
Traceback (most recent call last):
  File "c:\OctoPrint\venv\lib\site-packages\octoprint-1.3.0-py2.7.egg\octoprint\plugin\core.py", line 669, in _import_plugin
    instance = imp.load_module(key, f, filename, description)
  File "c:\OctoPrint\venv\lib\site-packages\octoprint_nautilus\__init__.py", line 14, in <module>
    from . import settings
ImportError: cannot import name settings
```

This is related to you OS (the awesome windows!). This doesn't happen on RPi or other *nix flavours. Unfortunately I don't have any windows machines to test, so it's gona take me a while...

Did you have this running previously on other versions of OctoPrint ? maybe RC ?

rberg11, 2016-12-09:
Unfortunately this is my first time trying octoprint. I had a spare intel nuc around and decided to give it ago. I appreciate the help and look forward to a fix and using your app.

Let me know if I can help you at all.

MoonshineSG, 2016-12-09:
At least I know it's not smth that has changed in version 3 versus RC. 
Just need to find a way to load that file... :D  (i now have a Virtual Machine to test on )

rberg11, 2016-12-09:
I'm running windows 10 x64 pro if it matters at all.

MoonshineSG, 2016-12-09:
can you edit the file `c:\OctoPrint\venv\lib\site-packages\octoprint_nautilus\__init__.py` and replace `from . import settings` with `from nautilus import settings` then try again ?

rberg11, 2016-12-09:
Yah I can give it a go in about 30 min. Have a print running at the moment.

rberg11, 2016-12-09:
Made the change, this is the new error.

2016-12-09 21:59:15,654 - py.warnings - WARNING - c:\OctoPrint\venv\lib\site-packages\octoprint-1.3.0-py2.7.egg\octoprint\plugin\core.py:1370: UserWarning: No PKG-INFO found for package: octoprint_nautilus
  warnings.warn('No PKG-INFO found for package: %s' % self.package_name)

2016-12-09 21:59:15,665 - octoprint.plugin.core - ERROR - Error loading plugin nautilus
Traceback (most recent call last):
  File "c:\OctoPrint\venv\lib\site-packages\octoprint-1.3.0-py2.7.egg\octoprint\plugin\core.py", line 669, in _import_plugin
    instance = imp.load_module(key, f, filename, description)
  File "c:\OctoPrint\venv\lib\site-packages\octoprint_nautilus\__init__.py", line 21, in <module>
    from socket import AF_INET, inet_pton
ImportError: cannot import name inet_pton

MoonshineSG, 2016-12-09:
It seams like that in windows the default implementation of sockets doesn't come with "inet_pton"... I'll need to look for alternative, or remove support for home/away in windows... 

But there is smth else I notice. Why do you have a "octoprint_nautilus_init.py"?

MoonshineSG, 2016-12-09:
Can you try the following:

run  `pip install win-inet-pton`

just before the line `from socket import AF_INET, inet_pton` add

```
import platform
if platform.system() == "Windows":
	import win_inet_pton
```

I have not tested this, but shouldn't work

rberg11, 2016-12-09:
Ill give it a try when i get home later, I have no clue why that file is there lol. I may try a reinstall from scratch and see what happens also.

Worst case scenerio ill reformat the nuc and install debian.

rberg11, 2016-12-10:
That code worked! Thanks!

MoonshineSG, 2016-12-10:
glad to hear is works. i'll put in the codes so it will work for future updates as well. 
thanks for testing.

closed 2016-12-10 by MoonshineSG