MoonshineSG/OctoPrint-Mobile #12: Turn Printer on and off

mrp192, 2016-12-10:
Hello,
 First I wanna say I love that there is an App I can use instead of having to do the web interface on my phone.

Now I have a relay setup that is controlled in octoprint to turn my printer on and off. It would be great if I was able to use that through the app! 
[image]

MoonshineSG, 2016-12-10:
assuming you use GPIO to control your relay, you can!!!!
but... you need to install another plugin, called "Octoprint Switch" .  See https://github.com/MoonshineSG/OctoPrint-Mobile/wiki/settings.ini#printer-commands-and-settings

Just a word of warning, the Switch plugin is not super well documented.

mrp192, 2016-12-10:
I do have GPIO installed. I have been doing it this way for awhile. I just didnt see an option in Nautilus.

MoonshineSG, 2016-12-10:
hahaha....the $0.99 is just to cover the Apple license (and a couple of beers) not to get rich :D 

glad you like it! Do leave a positive feedback on AppStore, as so far only people who don't like it - how can they!!! -  have left feedback ...

mrp192, 2016-12-10:
done.

MoonshineSG, 2016-12-10:
Let me know if you need help with "Swithc"

closed 2016-12-10 by MoonshineSG