MoonshineSG/OctoPrint-Mobile #14: Multiple printers - question

Radekva, 2016-12-12:
It is not issue but only question. It is able to work comfortably with two or more printers?

MoonshineSG, 2016-12-12:
Sadly, it doesn't and I can't see how this can be easily implemented.

pyjamasam, 2016-12-12:
I think adding a simple switching overlay. Like tabs or some other kinda simpler switching mechanism would do it. 

Really you just have to sent the web view to another URL. 

Loving the setting out of the settings app and making them in app would be a first step. You'll have more flexibility with respect to storing multiple instances of url's and API keys. 

But it just boils down to switching the base URL for all the calls.  Now your on a new printer. :-)

pyjamasam, 2016-12-12:
Cause I can't sort out how to edit a comment on my phone....

When I say switching overlay it's somthing that would be implemented in the objective-c code of the app. Not on the web side. 

Chris.

MoonshineSG, 2016-12-13:
There are a few other issues to be handled if multi printer needs to be supported, but nothing too complicated.
There is just a matter of finding the time for it...

Radekva, 2016-12-13:
So later it would support. Not in the near future?

MoonshineSG, 2016-12-13:
I am not saying it will. Not that it won't. Depends if there is interest (I'll leave this open so anyone can express interest). It would not make sense to develop it for one single users. 

I'll asses the effort and see if my time permits. It's also worth noting that I only have one printer (and I don't foresee buying/building another one in the near future (that is, if I wana stay married! :) )

closed 2017-01-09 by MoonshineSG