MoonshineSG/OctoPrint-Mobile #16: IOS App does not work still

macrhino, 2016-12-23:
As well. Configured. API key is there. There is a connection but nothing. There is a screen shot attached, 

The plugin is installed and shows up in Plugin Manager. 

[image]

MoonshineSG, 2016-12-23:
what do the logs show ? screen shot is from the iphone ?

macrhino, 2016-12-23:
Which logs? The plugin logs?

The URL is simple. When it is wrong, the program complains. Then the program notices that my API key was wrong (impossible if the URL was incorrect). I corrected the API key and I get a black screen everytime.

macrhino, 2016-12-24:
OK I know where the black screen comes from. It is the camera connected to the Octoprint. This is all that I get, just the camera.  No way to control anything or to make the camera picture go away.

MoonshineSG, 2016-12-25:
Oh.... ok. That happens when the connection is "external". What is the URL for your OctoPrint? Can you send the OctoPrint logs, to see why that happens ?

macrhino, 2016-12-25:
[octoprint.log.html.zip](https://github.com/MoonshineSG/OctoPrint-Mobile/files/672186/octoprint.log.html.zip)

MoonshineSG, 2016-12-26:
Merry Christmas!

By using a dynamic dns, you access your octoprint from the external network and therefor nautilus switches from full interface to pure webcam monitoring. 
I will add an option in the Nautilus settings to give the option to turn this function off, so that even from "outside" you will see the full interface. 

Give me a couple of days.

MoonshineSG, 2016-12-26:
@macrhino release 1.6.2 should solve your problem.

MoonshineSG, 2017-01-25:
Everything ok now?

closed 2017-01-29 by MoonshineSG