MoonshineSG/OctoPrint-Mobile #18: ios app still does not work - invalid api key

SirOTheOne, 2017-01-04:
i have followed the last posts, but the issue is not solved.

I also have the same problem - invalid API key

OctoPrint 1.3.0 (2016-12-08), Nautilus app id 1.3.3.440, Nautilus plug-in 1.6.2

i have copied and paste the api from OctoPi to Nautilus app (via web). I use octopi only local network

WHERE IS THE PROBLEM ???

Please solve it

MoonshineSG, 2017-01-06:
Some logs might help. You can try a fresh install. 

PS. NO NEED TO SHOUT!

SirOTheOne, 2017-01-06:
Hi
a fresh install of the app or the plugin?

MoonshineSG, 2017-01-06:
preferably a fresh install of everything. basically flash the SD card with a new image of octopi then install the plugin (update the API key in the iOS app) and try again.

There are only a couple of people who had this issue and I am not able to replicate it, nor find any explanation for it. The error is triggered by OctoPrint even before the plugin gets in action. The iOS app, **always**  sends the API key, so I am at lost to why this problem occurs to some people.

SirOTheOne, 2017-01-06:
now, i have installed everything new on a fresh formated sd card and also installed the app new.

the same as before: invalid API key

???

MoonshineSG, 2017-01-06:
can you tell me how if you machine setup ? in term of network configuration.

SirOTheOne, 2017-01-06:
i don´t know  what you mean exactly:
i´ve installed the octopi image on the sd card and then installed the nautilus plugin via octoprint with url from github.
i use octoprint only with a local ip (octopi.local), there is no route outside my local network.
the iphone is also connected local via wlan to octoprint

MoonshineSG, 2017-01-06:
thats what i needed to know. seams like a pretty straight forward setup. 
i assume you followed the API key setup steps from https://github.com/MoonshineSG/OctoPrint-Mobile/wiki/Install

SirOTheOne, 2017-01-06:
correctly. i follow exactly the procedure

SirOTheOne, 2017-01-06:
other plugins, like pushover, use also the api key and works without any issue

MoonshineSG, 2017-01-06:
no, pushover doesn't use the octoprint API key, but that's not the point.

I just installed a new SD card on a spare RPi, the plugin and a fresh copy of the iOS app, and no problems connecting... 

I will try to add some debug statements but it will take a while. once thats done, I'll ask you to do another test. as I said, I cannot replicate the error, so I can't really do any testing....

MoonshineSG, 2017-01-06:
can you open up your config.yaml and change "	
```
accessControl:
  enabled: true
``` 
to false and try again ?

SirOTheOne, 2017-01-06:
sorry, but the change does not change anything -> invalid API key

MoonshineSG, 2017-01-10:
trying to solve this...  Would you be willing to install a pre release test ios app ? If yes, you need to install "Test Flight" from Apple and then send me you id and I will send you an invitation to install the   app test version

SirOTheOne, 2017-01-10:
Hi Moonshine,

my apple id is: m.obermaier@mo-net.at

Michel

SirOTheOne, 2017-01-10:
@MoonshineSG

MoonshineSG, 2017-01-10:
I patched one file in OctoPrint files to add some debug statements. (see https://github.com/MoonshineSG/OctoPrint/commit/f85a4159d8159d3a96583e9127f72f2d90df4235)

On your RPi, If you can, replace the file `/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.0-py2.7.egg/octoprint/server/util/__init__.py` with `https://raw.githubusercontent.com/MoonshineSG/OctoPrint/f85a4159d8159d3a96583e9127f72f2d90df4235/src/octoprint/server/util/__init__.py`

Then if you don't have yet, create a file `/home/pi/.octoprint/logging.yaml` with the following content:

```
loggers:
  octoprint.apikey:
    level: DEBUG

```

restarts your OctoPrint service, try to connect with the iOS app, then send me the logs. 

There should be some `2017-01-10 17:47:59,423 - octoprint.apikey - DEBUG - request.headers.get('X-Api-Key') = .... ` entries in the log. Those are of interest.

SirOTheOne, 2017-01-10:
here is a part of the log file
[octoprint.log.zip](https://github.com/MoonshineSG/OctoPrint-Mobile/files/695828/octoprint.log.zip)

MoonshineSG, 2017-01-10:
and the api key is setup on the ios app ? 

the `haproxy` config is the standard ?

SirOTheOne, 2017-01-10:
yes, both are true.

SirOTheOne, 2017-01-10:
now its getting interesting: if i open octoprint on iphone browser, API key is shown as n/a and on safari on local machine it shows the normal API key ?!?!?!?

SirOTheOne, 2017-01-10:
now i tried it on my ipad. its the same message "invalid API key", but in the ipad browser i get the correct API key.

MoonshineSG, 2017-01-10:
what do you mean in the "ipad browser" ?

MoonshineSG, 2017-01-10:
how can the API be n/a and valid ? smth is really really wrong there...

SirOTheOne, 2017-01-10:
via safari browser on ipad

MoonshineSG, 2017-01-10:
oh... no via Nautilus you get erro message but on safari you can see the key on the octoprint settings ...

SirOTheOne, 2017-01-10:
correct

SirOTheOne, 2017-01-10:
but on both "invalid API key"

MoonshineSG, 2017-01-10:
"octoprint via safari browser on ipad: correct API key
octoprint via safari browser on iphone: n/a"

is the part I don't get...

SirOTheOne, 2017-01-10:
i also don´t get the part

i´m developing since about 30 years, but these undefinded issues i didn´t get any time

MoonshineSG, 2017-01-10:
About the same history here as well, and yes, software should not act up. 

Another slightly more complicated test would be to install a http header change extension (ModHeader) into chrome browser on a desktop/notebook, inject "API_KEY" change the User Agent to "Nautilus" and see what happens...

1n5aN1aC, 2017-01-10:
I could try this when my print finishes in several hours.

Should I also try changing the octoprint file to add logging, or do you think that unlikely to show anything?

MoonshineSG, 2017-01-10:
The OctoPrint file will show what actually gets to the server so yes, it would be usefull...

1n5aN1aC, 2017-01-13:
Alright, I made the changes, and I've got log entries & pictures for you!

```2017-01-13 05:05:34,682 - octoprint.apikey - DEBUG - <OctoPrintFlaskRequest 'http://home.13-thirtyseven.com/api/printerprofiles' [GET]>
2017-01-13 05:05:34,685 - octoprint.apikey - DEBUG - request.headers.get('X-Api-Key') = [None]
2017-01-13 05:05:34,686 - octoprint.apikey - DEBUG - apikey = [None]
2017-01-13 05:05:34,689 - octoprint.apikey - DEBUG - None
2017-01-13 05:05:34,697 - tornado.access - WARNING - 401 GET /api/printerprofiles (127.0.0.1) 23.46ms
2017-01-13 05:05:34,780 - octoprint.apikey - DEBUG - <OctoPrintFlaskRequest 'http://home.13-thirtyseven.com/plugin/nautilus/home' [GET]>
2017-01-13 05:05:34,782 - octoprint.apikey - DEBUG - request.headers.get('X-Api-Key') = [None]
2017-01-13 05:05:34,784 - octoprint.apikey - DEBUG - apikey = [None]
2017-01-13 05:05:34,787 - octoprint.apikey - DEBUG - None
2017-01-13 05:05:34,794 - tornado.access - WARNING - 401 GET /plugin/nautilus/home (127.0.0.1) 22.30ms
2017-01-13 05:06:07,996 - octoprint.apikey - DEBUG - <OctoPrintFlaskRequest 'http://home.13-thirtyseven.com/plugin/nautilus/home' [GET]>
2017-01-13 05:06:07,999 - octoprint.apikey - DEBUG - request.headers.get('X-Api-Key') = [None]
2017-01-13 05:06:08,001 - octoprint.apikey - DEBUG - apikey = [None]
2017-01-13 05:06:08,005 - octoprint.apikey - DEBUG - None
2017-01-13 05:06:08,016 - tornado.access - WARNING - 401 GET /plugin/nautilus/home (127.0.0.1) 31.04ms
```

Here are screenshots from my phone of what I see:

[image]
[image]

MoonshineSG, 2017-01-13:
what iPhone are you using ? iOS version ? 

it seams to me that somehow the header set by the app gets removed somewhere between the phone and the octoprint server. Using chrome (with User agent and APIKEY header injection) i was able to connect to your printer (!!!! - change your API key NOW! )

SirOTheOne, 2017-01-13:
i´m using an iphone 6 with ios 10.2 and an ipad mini with ios 9.3.5

on both devices it´s the same "invalid API key"

1n5aN1aC, 2017-01-13:
Whoops, you are right, I forgot to mask my IP from the logs.  >.<   Well, at least we learned something from it....

I am on iPhone 5s, iOS version 9.3.3 (And no, I will not be upgrading, sorry....) so I suppose there is some possibility that may play a factor into it, but it seems unlikely since @SirOTheOne is on 10.x

What I can't understand is why I have had it SOMETIMES work for brief periods, before it stops working...  :/

MoonshineSG, 2017-01-13:
No, don't think it's a problem with the iOS... 
I really suspect some of the network equipment... but I am not network engineer...
Is it possible to try with local address? 
Or... do you know how to configure the RPi as hotspot?

SirOTheOne, 2017-01-13:
I don´t think that this is an issue of networking.

I use slic3r, which uses API key, too and it works without any problem and i´m working on a local network

1n5aN1aC, 2017-01-13:
I believe I have accessed and used it before via my local 10.0.0.x address, and could not get it to reliably work, but I will try it again soon.

1n5aN1aC, 2017-01-13:
Hmmm, not acting any different for me as of now:

```2017-01-13 06:33:24,373 - octoprint.apikey - DEBUG - <OctoPrintFlaskRequest 'http://10.0.0.36/api/printerprofiles' [GET]>
2017-01-13 06:33:24,377 - octoprint.apikey - DEBUG - request.headers.get('X-Api-Key') = [None]
2017-01-13 06:33:24,378 - octoprint.apikey - DEBUG - apikey = [None]
2017-01-13 06:33:24,382 - octoprint.apikey - DEBUG - None
2017-01-13 06:33:24,392 - tornado.access - WARNING - 401 GET /api/printerprofiles (127.0.0.1) 26.27ms
2017-01-13 06:33:24,469 - octoprint.apikey - DEBUG - <OctoPrintFlaskRequest 'http://10.0.0.36/plugin/nautilus/home' [GET]>
2017-01-13 06:33:24,472 - octoprint.apikey - DEBUG - request.headers.get('X-Api-Key') = [None]
2017-01-13 06:33:24,473 - octoprint.apikey - DEBUG - apikey = [None]
2017-01-13 06:33:24,477 - octoprint.apikey - DEBUG - None
2017-01-13 06:33:24,486 - tornado.access - WARNING - 401 GET /plugin/nautilus/home (127.0.0.1) 24.95ms
```

If I have time tomorrow (questionable)  I might even grab a packet capture to see what's coming from the app in terms of headers.  :)

1n5aN1aC, 2017-01-13:
Alright, I've fired up wireshark on my desktop computer, as well as arp poisoning between my phone and the octoprint server.  This way I am seeing all network packets transferred between my phone and the server.

Here is an example of just the HTTP header portion of one of the packets I captured:

```
Hypertext Transfer Protocol
    GET /plugin/nautilus/home HTTP/1.1\r\n
    Host: 10.0.0.36\r\n
    Content-Type: application/json\r\n
    Accept-Encoding: gzip, deflate\r\n
    Cookie: <removed for privacy>
    Connection: keep-alive\r\n
    X-Api-Key: None\r\n
    Accept: */*\r\n
    User-Agent: Nautilus/1.3.3.440\r\n
    Referer: http://10.0.0.36/\r\n
    Accept-Language: en-us\r\n
    X-Requested-With: XMLHttpRequest\r\n
    \r\n
    [Full request URI: http://10.0.0.36/plugin/nautilus/home]
    [HTTP request 1/4]
    [Response in frame: 1929]
    [Next request in frame: 1966]

```

In my mind (unless I'm misunderstanding something)  this pretty much shows the problem is with the app not always sending the API-KEY.

HOWEVER... This is where my expertise ends, and I have no idea how to poke at anything in apps, and I have no experience with how they are coded or what could be the problem.  Hopefully, this will help you look in the right places though.

MoonshineSG, 2017-01-13:
the key is sent by the iOS app on the first previous request ( for http://ocoptint.local/ as API_KEY then gets; picket up by JavaScript and injected in all the JQuery requests. It seams that the header "API_KEY" is dropped. The new iOS app (pending apple review) is sending both API_KEY and X-Api-Key then the JS (which will need to be updated) will pick up the one that has value (need to keep the old one for backwards compatibility)

Hopefully this will help. Give it a few more day. Apple is busy after holidays with approvals..

SirOTheOne, 2017-01-13:
@MoonshineSG please send me the id for apple test flight to check to new version (see above)

MoonshineSG, 2017-01-13:
if you installed "Test Flight" app from Apple, login with the ID you sent me, once the Nautilus is approved for testing, you will receive a notification to install the test version. 

You will then need to manually update one file of your Nautilus plugin to the version on the master github (unreleased) https://raw.githubusercontent.com/MoonshineSG/OctoPrint-Mobile/master/octoprint_nautilus/templates/nautilus_index.jinja2 to replace the file in `/home/pi/oprint/lib/python2.7/site-packages/octoprint_nautilus/templates`

SirOTheOne, 2017-01-13:
i haven´t sent you an apple id. You should send me an email to the above address and i will send you my apple id.

MoonshineSG, 2017-01-13:
@SirOTheOne i see... it's not the "m.o.........@......at" ?

SirOTheOne, 2017-01-13:
yes, it is the m.o.........@......at, but i need the inviting code.

MoonshineSG, 2017-01-13:
that will come once the app is approved (even test apps need to go through apple approval process)

SirOTheOne, 2017-01-13:
ok, thanks. i have never used test flight. i´m testing on my own devices.

MoonshineSG, 2017-01-13:
Even for own tests is usefull to test the deployment. For "internal" users there's no approval process. "External" it's a pain :)

SirOTheOne, 2017-01-13:
that´s true - apple approval process !!!!! i know it from providing apps -> painful

SirOTheOne, 2017-01-14:
Hi,
now i have tested on 2 devices (iPhone 6-10.2, iPad-9.3.5) and on both devices it´s the same "imvalid API key"
[part of logging.zip](https://github.com/MoonshineSG/OctoPrint-Mobile/files/705779/part.of.logging.zip)

MoonshineSG, 2017-01-14:
you updated the plugin? the template from my repo. you can uninstall your plugin and then reinstall via the plugin manager

SirOTheOne, 2017-01-14:
i have updated your plugin

1n5aN1aC, 2017-01-14:
I received your Test Flite invite, but I cannot currently replace the file on the printer, since I have a very long print running.  I'll get back to you once I'm able to do that.  :P

MoonshineSG, 2017-01-14:
@SirOTheOne I've added some debug statements in the plugin. please update again.

SirOTheOne, 2017-01-14:
the plugin is uptodate

MoonshineSG, 2017-01-14:
then try to connect again and send me the log file  (remember to restarts octoprint after a plugin update)

SirOTheOne, 2017-01-14:
2017-01-14 06:16:43,554 - octoprint.apikey - DEBUG - request.headers.get('X-Api-Key') = [616EB6EC46004E6C872765E5C968FE13]
2017-01-14 06:16:43,555 - octoprint.apikey - DEBUG - apikey = [616EB6EC46004E6C872765E5C968FE13]
2017-01-14 06:16:43,566 - octoprint.apikey - DEBUG - <OctoPrintFlaskRequest 'http://octopi.local./api/system/commands' [GET]>
2017-01-14 06:16:43,567 - octoprint.apikey - DEBUG - request.headers.get('X-Api-Key') = [616EB6EC46004E6C872765E5C968FE13]
2017-01-14 06:16:43,568 - octoprint.apikey - DEBUG - apikey = [616EB6EC46004E6C872765E5C968FE13]
2017-01-14 06:16:43,569 - octoprint.apikey - DEBUG - request.headers.get('X-Api-Key') = [616EB6EC46004E6C872765E5C968FE13]
2017-01-14 06:17:26,858 - octoprint.plugins.nautilus - DEBUG - will_handle_ui request headers:
2017-01-14 06:17:26,859 - octoprint.plugins.nautilus - DEBUG - X-Api-Key: EC72EA4B16A747A2A35AD0FB49A44081
Api-Key: EC72EA4B16A747A2A35AD0FB49A44081
Host: octopi.local
Connection: close
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8
User-Agent: Nautilus/1.3.4.444
Accept-Language: de-at
X-Forwarded-For: 192.168.1.206
Accept-Encoding: gzip, deflate

2017-01-14 06:17:27,890 - octoprint.apikey - DEBUG - <OctoPrintFlaskRequest 'http://octopi.local/api/printerprofiles' [GET]>
2017-01-14 06:17:27,891 - octoprint.apikey - DEBUG - request.headers.get('X-Api-Key') = [None]
2017-01-14 06:17:27,892 - octoprint.apikey - DEBUG - apikey = [None]
2017-01-14 06:17:27,893 - octoprint.apikey - DEBUG - None
2017-01-14 06:17:27,897 - tornado.access - WARNING - 401 GET /api/printerprofiles (127.0.0.1) 11.75ms
2017-01-14 06:17:28,118 - octoprint.apikey - DEBUG - <OctoPrintFlaskRequest 'http://octopi.local/plugin/nautilus/home' [GET]>
2017-01-14 06:17:28,119 - octoprint.apikey - DEBUG - request.headers.get('X-Api-Key') = [None]
2017-01-14 06:17:28,119 - octoprint.apikey - DEBUG - apikey = [None]
2017-01-14 06:17:28,120 - octoprint.apikey - DEBUG - None
2017-01-14 06:17:28,125 - tornado.access - WARNING - 401 GET /plugin/nautilus/home (127.0.0.1) 11.16ms
2017-01-14 06:17:32,443 - octoprint.apikey - DEBUG - <OctoPrintFlaskRequest 'http://octopi.local/plugin/nautilus/home' [GET]>
2017-01-14 06:17:32,445 - octoprint.apikey - DEBUG - request.headers.get('X-Api-Key') = [None]
2017-01-14 06:17:32,451 - octoprint.apikey - DEBUG - apikey = [None]
2017-01-14 06:17:32,454 - octoprint.apikey - DEBUG - None
2017-01-14 06:17:32,462 - tornado.access - WARNING - 401 GET /plugin/nautilus/home (127.0.0.1) 27.34ms

MoonshineSG, 2017-01-14:
Somehow your http header gets modified from "API_KEY" to "Api-Key". Even then, the new template should pick up the "X-Api-Key" which disappears... Mystery.  

Something  I notice, is the some requests go to  `http://octopi.local./` with a `.` at the end. 
Can you check your URLS ? Check HAProxy config file, and your Nautilus setting.

[95 more replies]

closed 2017-01-29 by MoonshineSG