MoonshineSG/OctoPrint-Mobile #19: Mobile app does not work

clarkewd, 2017-01-18:
Downloaded app from iTunes - but it's stuck after opening it on iPhone.

> Before you can use this application, the "Nautilus" plugin needs to be installed on OctoPrint and you need to setup the OctoPrint server URL and the API KEY in the settings of this app.

Screenshot:

[image]

How do I get to the "settings" ?

1n5aN1aC, 2017-01-18:
You go to the 'settings' app on your phone.

Scroll down near the bottom, and there will be a link 'nautilus'

closed 2017-01-18 by clarkewd

MoonshineSG, 2017-01-19:
I guess you found it. Unfortunately, the end of the message did not fit into the screen.  It reads "(shake to load the app settings)".... :D 
I'll change the font size in the next release.