MoonshineSG/OctoPrint-Mobile #20: Plugins conflict

louvanna, 2017-01-21:
I had configured the timelapse camera with a webcam with success. This included the config for adding controls in the page. Later, I added the Email Notification plugin, which works as expected. But it has defeated the functionality of the timelapse.

Any guidance to resolve the plugin conflict is appreciated.

MoonshineSG, 2017-01-21:
sorry, but how is this related to the Nautilus plugin ?

MoonshineSG, 2017-01-21:
please explain in detail. what is the conflict? logs and versions

louvanna, 2017-01-21:
OctoPrint:  Version: 1.3.0 (master branch) 
Email-Notifier installed with plugin facility.
mjpeg-streamer, after the install of Email-Notifier, is started at start-up with a daemon webcamd, whch I found at /root/bin/webcamd.  
The timelapse daemon, at /home/pi/scripts/webcamDaemon,  when run with ./webcam start. does not show the error.
pi@octopi:~/scripts $ ./webcam start
./webcam: started

But directly running webcamDaemon, reveals, ( set -x ) added
 ---------- snip ------------
MJPG Streamer Version.: 2.0
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: -1
 i: Format............: YUYV
 i: JPEG Quality......: 80
 i: TV-Norm...........: DEFAULT
libv4l2: error setting pixformat: Device or resource busy
Unable to set format: 1448695129 res: 640x480
 Init v4L2 failed !! exit fatal 
 i: init_VideoIn failed
+ popd
~/scripts
+ sleep 120
------- end snip --------------

The complaint is correct since the camera is busy by the 'other' daemon.
Thereafter, the timelapse when started errors since the /dev/video0 is 'busy' and errors out.

Other details needed ?

Louis

MoonshineSG, 2017-01-22:
now i really have to ask how is this related to Nautilus ??

louvanna, 2017-01-22:
Apologies, I now see that this is OctoPrint-"mobile", upon closer look. My usage is the OctoPrint implemented upon a printer and not the mobile aspects. 
For the record, I have resolved the conflicting installations by not having each of 1) the native timelapse and 2) plugin Email-notifier,  invoke separate mjpeg-streamer daemons. One instance is all that is needed for each to perform the necessary image captures.

again, apologies.

closed 2017-01-22 by louvanna