MoonshineSG/OctoPrint-Mobile #22: IOS app is asking for new plugin 1.7

pmbroth, 2017-01-24:
Hello,

I have installed the latest plugin - 1.6.5.  The IOS app is wanting new plug in 1.7.  I cannot find it anywhere.

Thank you

Brian

DaveHerm, 2017-01-24:
Hi,

the same for me, do get the message to update to 1.7 but the latest release is 1.6.5
please upload the new release

br
thomas

MoonshineSG, 2017-01-25:
iOS app release was , by mistake set to auto instead of manual, so once approved it was published.
The plugin was all prepared but not released and I had to fly off (literally)

I re-coded a quick fix of version 1.7 on my iPhone while transferring from airport to hotel :) another rushed job. Please let me know if it works.

pmbroth, 2017-01-25:
That fixed it thank you!

closed 2017-01-25 by pmbroth

DaveHerm, 2017-01-25:
can approve that it works for me too

thank you

MoonshineSG, 2017-01-25:
Glad to hear it's ok. 

Leave a positive review on AppStore if you enjoy the plugin. Thanks.