MoonshineSG/OctoPrint-Mobile #23: IOS App and the plugin Detailed Progress

DaveHerm, 2017-01-25:
If you have the plugin "Detailed Progress" installed which changes the info on the Display every 10 seconds.

You get all the messages as popup in the app.
is there a possibilty to supress these or at least just show one so you dont have to aknowledge every one.
Reason if u happen to print more then 10 Minutes and have the app running you get 6 messages/minute :)

my workaround for now will be to disable the plugin, but if by any chance you ahve the time, feel free to help me out :)

br
Thomas

MoonshineSG, 2017-01-25:
I am not familiar with that plugin but I will have a look. 6 messages a min would annoy me as well :)

But I need a while as I am on holiday right now and only return by mid Feb.

NeoTrace82, 2017-01-25:
sure take your time :)

i will disable the plugin in the meanwhile

Appreciate your offer!

tfohlmeister, 2017-01-28:
I am having the exact same problem with the app. Maybe you could add a setting to disable those popups altogether?

MoonshineSG, 2017-01-28:
The plan is to Ignore certain messages or add a generic "disable m117 messages" setting

MoonshineSG, 2017-02-04:
@NeoTrace82 , @tfohlmeister  release 1.7.1 should solve this. 
Nautilus detects the existence of "Detailed Progress" and automatically disables the popups. There is also a new setting that allows you to manually disable to popups (just in case)

Hope this helps.

NeoTrace82, 2017-02-04:
great stuff :)

one thing that i did notice the system commands (shutdown/reboot) i did add 2 new commands to turn on and off my printer with wireless plug sockets can they be shown too?

to be sure you get what i ment here is the tutorial link i have written
https://hackaday.io/project/19507-octoprint-mod-to-remotly-trigger-plug-sockets

tfohlmeister, 2017-02-04:
Great! Detection of "Detailed Progress" seems to work fine (no need to set the general option). Thank you!

MoonshineSG, 2017-02-05:
@NeoTrace82 I have an additional plugin (https://github.com/MoonshineSG/OctoPrint-Switch) (semi integrated with Nautilus) that uses GPIO to turn on/off power. Maybe with some changes it could be used in your case.  Do you have a way to check if power is on or off ?

MoonshineSG, 2017-02-05:
@NeoTrace82 see https://github.com/MoonshineSG/OctoPrint-Mobile/wiki/settings.ini#printer-commands-and-settings

NeoTrace82, 2017-02-05:
Im sending commands as I trigger it via 433mhz sender

So the plugin of yours cannot be taken

MoonshineSG, 2017-02-05:
Not as it is. But with some changes....maybe... 
Or maybe smth else... let me think about it

MoonshineSG, 2017-02-05:
@NeoTrace82 
I understand that the On/Off commands are:
Power On: sudo /home/pi/raspberry-remote/send 11110 1 1
Power Off: sudo /home/pi/raspberry-remote/send 11110 1 0

is there a way to check if power is currently ON or OFF ?

NeoTrace82, 2017-02-05:
It just sends the command no verification or anything 

So it works the same as the system off command like in octoprint itself

MoonshineSG, 2017-02-05:
i see... in that case there will always be 2 buttons, regardless of the current state of the power. (just like in the menu)

MoonshineSG, 2017-02-05:
@NeoTrace82 : you can try a test version on the plugin. 
Uninstall the plugin via the Plugin Manager and then using the url (https://github.com/MoonshineSG/OctoPrint-Mobile/archive/custom_power_buttons.zip) install it again. 

Then, in the `config.yaml` you need to rename your system actions `power_on_printer` and `shutdown_printer` and Nautilus will pick them up and create the buttons

[image]

Note: this only works if you **don't** have the Switch plugin enabled.

MoonshineSG, 2017-02-07:
If all is good, the changes for "custom power buttons" will be in release 1.8 with other enhancements and bug fixes.

NeoTrace82, 2017-02-07:
it does work like a charm :)

thank you

MoonshineSG, 2017-02-07:
great. I'll go ahead and do release it.

closed 2017-02-07 by MoonshineSG