MoonshineSG/OctoPrint-Mobile #24: I cant get this to work with my printer

Raymosj, 2017-01-25:
I can't get past the screen that says shake to load url and api...i have the correct info in the two spots. I heard there was a bug in the software with IOS that wanted the plugin program on Octoprint to be 1.7. I see version 1.7 plugin now...  I deleted your octoprint plugin and downloaded the 1.7 version and still no luck.... What could be causing me to not get by this screen.

Thanks for the help....

MoonshineSG, 2017-01-25:
Did you setup the URL and the api key in the application settings?

dshapdesign, 2017-01-25:
Where is the 1.7 version? I only see the 1.6.5 listed here.

Raymosj, 2017-01-25:
Yes when your application wasn't working I bought Octo remote from the App Store and used the same settings on their program and it works perfectly. But their program is so rudimentary. I really want to use yours. I just can't get past that screen that asks for URL and API key. What can be the issue.

Raymosj, 2017-01-25:
The 1.7 version is on Octoprint plug-in.

dshapdesign, 2017-01-25:
Yes. I have downloaded the octoprint 1.6.5 from the plugin manager in octoprint. WHERE CAN I DOWNLOAD 1.7?

Raymosj, 2017-01-25:
I'm in the car driving I'll be home in a few minutes

Raymosj, 2017-01-25:
We did it from our octal Pratt Internet connection in the ad plug-ins section. I want to settings scroll down to plug-ins went to the bottom and more Addmore and was the second or third print pages the octal print 1.7 plug-in

Raymosj, 2017-01-25:
Then I hooked up my computer to my 3-D printer and connected againand re-downloaded your iOS app and I can't get past that screen that wants me to add settings

dshapdesign, 2017-01-25:
[image]
It doesn't even show 1.7 in the release notes... Am I missing something here? I downloaded the octoprint add-on on monday and was only given 1.6.5 as an option to download. I even had it search for updates and I got nothing.

dshapdesign, 2017-01-25:
Sorry, I downloaded it yesterday not Monday.

Raymosj, 2017-01-26:
I'm back at home now give me one second and I'll take a picture of it on my tablet OK

Raymosj, 2017-01-26:
I'm sorry what I meant was not a less 1.7. One of the guys on the forum for formbot had put a service request on your same site I did three or four days ago. And you must answered it and said the iOS program for novelist was calling for the 1.7 nautilist plug-in and it didn't exist. You supposedly fixed it. I bought the product at the same time this was going on. The guy on the form told me this morning it was fixed. But I still can't get it going.

[image1.JPG]

MoonshineSG, 2017-01-26:
I didn't have time to write the release notes. But the release is there

dshapdesign, 2017-01-26:
Cool. What is the url for the update? Yesterday octoprint thought that 1.6.5 was the latest version.

Raymosj, 2017-01-26:
so im looking at your program now on iphone....Yahooooooo....Now its working.......

Raymosj, 2017-01-26:
thanks I dont know why....but its up....VERY NICE Ill give you a great rating on apple

Raymosj, 2017-01-26:
wow best IOS controller for printer to date!!!!!

closed 2017-01-26 by Raymosj

MoonshineSG, 2017-01-26:
Glad you got it working. 
Positive rating would be much appreciated. So far only people who didn't manage to have it working (some bugs ,true ) gave nasty comments and low rating