MoonshineSG/OctoPrint-Mobile #25: How do I enable Octoprint Switch in my code?

Raymosj, 2017-01-26:
Hi There,

I love your program. I wanted to know how I add octoprint switch to Nautilus? I really want that functionality. I'm clueless after reading the octoprint switch github site. DO I need to add code? What code? Where? thatnls so much for the great support.....

1n5aN1aC, 2017-01-26:
No, you should not need to edit code yourself...

You will need to install the separate "Octoprint-Switch" plugin in your Octoprint, then edit the config and tell it which pins you have connected the printers power to, lights, etc.

Raymosj, 2017-01-26:
is there any instructions on doing it anywhere?

1n5aN1aC, 2017-01-26:
Follow the instructions [here](https://github.com/MoonshineSG/OctoPrint-Mobile/wiki/Install), but instead of using that URL, use the one for [Octoprint-Switch](https://github.com/MoonshineSG/OctoPrint-Switch) which is ```https://github.com/MoonshineSG/Octoprint-Switch/archive/master.zip```

Once it's installed, the readme [on the Octoprint-Switch page](https://github.com/MoonshineSG/OctoPrint-Switch) explains the configuration options.

Raymosj, 2017-01-26:
thanks so much.....

MoonshineSG, 2017-01-26:
@1n5aN1aC is right. There should be no need to edit code. 

@1n5aN1aC, just curious.. are you using the Switch as well?

1n5aN1aC, 2017-01-26:
Not yet, but I plan to when I have the chance to set it up again.

MoonshineSG, 2017-02-02:
Once you enable the Switch plugin, the 4 additional functions will be available on OctoPrint as well as Nautilus.

closed 2017-02-06 by MoonshineSG