MoonshineSG/OctoPrint-Mobile #26: How to access Nautilus via NAT/port forwarding

beercity, 2017-02-01:
I am having issues figuring out how to port forward Nautilus.  I assumed it ran on the same port as Octoprint (80).  I set up a forward on an obscure outside port to port 80 on the inside, but this did not work.  I changed to forward 80 on the outside to 80 on the inside and this did not work either.

Does Nautilus use its own TCP port for connection to Octoprint?

MoonshineSG, 2017-02-01:
No. As long as you can access OctoPrint  via a browser,  Nautilus should be ok as well.

beercity, 2017-02-01:
Does the app support url:port?  Or can it only access it on the default port?
It does not seem to work either way, but I can copy/paste that URL into Safari and Octoprint comes right up.  If I put the internal IP in the settings while on inside WiFi, it works, so I am confident that my API key is correct.

MoonshineSG, 2017-02-01:
URL:port should work just fine.
What's the error message?

MoonshineSG, 2017-02-01:
Just guessing here... you don't have a web camera on your printer...

If that's the case, please go under Nautilus plugin settings (on OctoPrint settings, not iOS) and make sure "External access show only webcam" is not checked.

beercity, 2017-02-02:
Awesome that appears to have done the trick!  Thank you!

closed 2017-02-02 by beercity

MoonshineSG, 2017-02-02:
Awesome. I'll make sure a more detailed error message is shown when camera is not available. 

PS: if you enjoy the plugin, an Apple Store positive review is appreciated and please spread the word. Tx.