MoonshineSG/OctoPrint-Mobile #29: Axes reversal

MoonshineSG, 2017-02-06:
@SirOTheOne 
Trying to understand what is the behaviour you want to get. 

My understanding: only the "relative" movement has problems. That would be  all the red/blue buttons. Is that correct?

SirOTheOne, 2017-02-06:
@MoonshineSG 
All movements with the red/blue buttons are vice verca (X, Y and Z). in octoprint the movements are correct.

SirOTheOne, 2017-02-06:
all other movements can be set as gcode in the config file

MoonshineSG, 2017-02-06:
I see. So on your OctoPrint printer profile, you have all the axes "inverted" ? 

I have a test version of the plugin. You need to remove the current plugin, and install with this custom url (https://github.com/MoonshineSG/OctoPrint-Mobile/archive/invert_axis.zip) 

No need to configure anything, it picks up the "invert" from the OctoPrint profile. 

It's not 100% done, but it should work. Let me know how it goes and whats missing.

SirOTheOne, 2017-02-06:
now i have installed it, first deinstalled the plugin and reboot pi, installed the archive and reboot pi
now the plugin doesn´t work anymore. the ios app "says" Loading... and does not stop or connect to pi

SirOTheOne, 2017-02-06:
i also rebooted the ios device. connection via web is established

MoonshineSG, 2017-02-06:
can you check the logs ?

MoonshineSG, 2017-02-06:
btw, theres no need to reboot pi or the ios. Restart OctoPrint and the iOS app will do.

SirOTheOne, 2017-02-06:
sorry, there was any network config issue on my side, but red/blue buttons 1/10/100 work correct, max./min x/y/z are still vice verca

MoonshineSG, 2017-02-06:
oh... those are not changed. I only changed the relative movements. 

so now, say on X, left, should go to "max" ? maybe those can be changed via the config ?

SirOTheOne, 2017-02-06:
and i think there is an issue between ios between ios 10.2.1 (my iphone) and ios 9.3.5 (my ipad), because on the iphone the webapp works, on the ipad only the messages appears loading. both device are on the same network and establish the webapp on the same hostname

MoonshineSG, 2017-02-06:
can you check if youre ipad app is the latest version ?

SirOTheOne, 2017-02-06:
X left should go to max, but it goes to min (right)

x right move to max.X on printer

ios version 1.4.503 on both devices

MoonshineSG, 2017-02-06:
that can be changed in the config
```
goto_x_max =
goto_y_max =
goto_z_max =
 
goto_x_min = 
goto_y_min = 
goto_z_min = 
```

the ipad issue i am not sure. now you can enable "debug" in the plugin settings and then check the logs...

SirOTheOne, 2017-02-06:
log from ipad:
2017-02-06 12:24:44,357 - octoprint.plugins.nautilus - DEBUG - on_ui_render request headers:
2017-02-06 12:24:44,357 - octoprint.plugins.nautilus - DEBUG - Pragma: no-cache
Cookie: session_P80=.......
Host: octopi.local
Connection: close
Cache-Control: no-cache
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8
User-Agent: Nautilus/1.4.503
Accept-Language: de-at
X-Forwarded-For: 192.168.1.206
Accept-Encoding: gzip, deflate

SirOTheOne, 2017-02-06:
ans from iphone:
2017-02-06 12:26:25,325 - octoprint.plugins.nautilus - DEBUG - X-Forwarded-For : [192.168.1.220]
2017-02-06 12:26:25,327 - octoprint.plugins.nautilus - DEBUG - Forced 'home' access...
2017-02-06 12:26:25,456 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.1.220
2017-02-06 12:26:25,595 - octoprint.plugins.nautilus - DEBUG - Can't read the hotend config file. Default values used.
2017-02-06 12:26:25,636 - octoprint.plugins.nautilus - DEBUG - will_handle_ui request headers:
2017-02-06 12:26:25,637 - octoprint.plugins.nautilus - DEBUG - Referer: http://octopi.local/
Host: octopi.local
Connection: close
Cache-Control: max-age=0
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8
User-Agent: Nautilus/1.4.503
Cookie: session_P80=....
Accept-Language: de-at
X-Forwarded-For: 192.168.1.220
Upgrade-Insecure-Requests: 1
Accept-Encoding: gzip, deflate

2017-02-06 12:26:25,829 - octoprint.server.util.sockjs - INFO - Client connection closed: 192.168.1.220
2017-02-06 12:26:26,302 - octoprint.plugins.nautilus - DEBUG - X-Forwarded-For : [192.168.1.220]
2017-02-06 12:26:26,303 - octoprint.plugins.nautilus - DEBUG - Forced 'home' access...
2017-02-06 12:26:26,421 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.1.220
2017-02-06 12:26:26,546 - octoprint.plugins.nautilus - DEBUG - Can't read the hotend config file. Default values used.

MoonshineSG, 2017-02-06:
Did the iPad ever worked? 
I'll setup a 9.3.5 simulator to test, but it should not make a difference. 
Update iOS is an option?

SirOTheOne, 2017-02-06:
ipad works with your last version 1.7

it´s an ipad mini (1st), update no option

MoonshineSG, 2017-02-06:
Ok. I have my wife's iPad mini with 9.3.3 and I can see your problem. I'll investigate and get back to you.

MoonshineSG, 2017-02-06:
new version should fix the problem. you can reinstall (using the same url) or manually patch one single file based on the latest commit (https://github.com/MoonshineSG/OctoPrint-Mobile/commit/5664b9d46c7b4cf2d4ab4ec371273ce084812b67)

SirOTheOne, 2017-02-06:
now it works on both devices, thanks

MoonshineSG, 2017-02-06:
great. so the inverted axes work well ?

SirOTheOne, 2017-02-06:
perfect, everything works like it should

maybe feature request: is it possibly to show the current coords of the hotend in the movement screen?

MoonshineSG, 2017-02-06:
sorry, that is very taxing on the server. it would require constant pooling and that's not something you want. I don't think i will be implementing it. 

a "on demand" show of XYZ is possible. just like the way "current Z" is updated when the "ofset" panel is activated. it requires a M114 to be sent to the printer and result to be parsed, but it is possible. Since is on the movement, it would be inactive during printing.

would that be useful ?

SirOTheOne, 2017-02-06:
as you wrote constant pooling is not needed (server brake down), on demand would be very useful

during printing inactive, parse M114 is a good try

MoonshineSG, 2017-02-07:
@SirOTheOne just curious... why did you choose to have your "home" back-right not front-left ?

SirOTheOne, 2017-02-07:
due to to the printer construction it is better when home is in back/right for removing the printed object  from the bed

MoonshineSG, 2017-02-07:
Interesting. Must be a special build. 

The "inverted axes" changes will be in the 1.8 release, together with other enhancements and bug fixes.

closed 2017-02-07 by MoonshineSG