MoonshineSG/OctoPrint-Mobile #31: Old version problem with IOS app

jotaperamez, 2017-02-23:
I was checking Nautilus app 1.9 and when i tried to start it, it show me a message "Your version of Nautilus is too older". Some suggestion?

thanks

MoonshineSG, 2017-02-23:
Please read https://github.com/MoonshineSG/OctoPrint-Mobile/wiki/Update

jotaperamez, 2017-02-23:
i did, my version of Nautilus is 1.9 but same problem

MoonshineSG, 2017-02-23:
1. check what does `http://YOUR_URL_HERE/plugin/nautilus/static/js/version.js` shows. If it does not show 1.9, try to manually re-install the plugin.

2. try to "kill" the iOS app and start it again.  (cache? - shouldn't be)

3. what OS are you running OctoPrint on ?

jotaperamez, 2017-02-24:
just reinstall iOS app and works! :) ty

closed 2017-02-24 by jotaperamez

MoonshineSG, 2017-02-24:
strange. but glad it works. enjoy.