MoonshineSG/OctoPrint-Mobile #35: http-auth prompts/responses and user agent strings

dhardyuk, 2017-03-20:
Hi there,

Many thanks for your work on a project which works beautifully for most of my use case :-)

I have recently followed a how-to to enable http-auth challenge/security on haproxy for octopi. I've also added in an IP rule which only requires haproxy to prompt for credentials when the client is not connecting from my lan. The setup works well and is prompting for credentials as expected when i use safari on the iphone.

Everything works fine for Nautilus on my iphone when I'm on the lan but when I turn wifi off and connect by 3g the expected http-auth challenge/prompt isn't pushed to the ui so the Nautilus app just shows the blue loading screen.

I am running a fully patched and up to date Octopi image with everything (including all plugins and pip) updated as of yesterday with a fully qualified domain name that resolves on the internet to the external ip of my router and resolves internally to the ip of the Octopi installation.

Is it possible to enable this kind of security for the Nautilus IOS client? if not, is there an http header/user agent string in the client that I can leverage to 'exempt' Nautilus from the http-auth requirement?

I'm also looking at on demand VPN access to the lan from the iphone which will be an adequate work-around once if I can get it working :-)

Regards

David

MoonshineSG, 2017-03-21:
I'm glad you enjoy it!

Unfortunately, http-auth is not something I am looking at implementing. But you can of course skip that for Nautilus using `acl` with `hdr_beg(User-Agent) -i nautilus`

The User-Agent sent by Nautilus is `Nautilus X` where X represents the version, and currently is `1.4.1.512`

Hope this helps.

PS. and remember, you will need to do port forwarding on your router as well to allow external access to your printer.

dhardyuk, 2017-03-21:
That's fantastic :-)

For anyone that's interested here is the working haproxy.cfg file I'm using with # symbols where you can put your own data. This is working with ports 80 and 443 being forwarded to the internal ip of the PI (I've got a dhcp reservation for my pi and the router has a static DNS overide entry for the fully qualified domain name of octopi.#mydomain.#com). I've followed my nose through some how-to's to get letsencrypt running on the Pi and have fettled the haproxy.cfg file (with no eye for elegance so there is some duplication) to get this:

global
        maxconn 4096
        user haproxy
        group haproxy
        log 127.0.0.1 local1 debug

defaults
        log global
        mode http
        option httplog
        option dontlognull
        retries 3
        option redispatch
        option http-server-close
        option forwardfor except 127.0.0.1
        maxconn 2000
        timeout connect 5s
        timeout client 15min
        timeout server 15min

frontend public
        bind *:80
        bind 0.0.0.0:443 ssl crt /#etc#/#haproxy#/#certs#/#snakeoil#.pem
        option forwardfor except 127.0.0.1
	use_backend webcam if { path_beg /webcam/ }
	acl letsencrypt-acl path_beg /.well-known/acme-challenge/
	use_backend letsencrypt-backend if letsencrypt-acl
        default_backend octoprint
        errorfile 503 /etc/haproxy/errors/503-no-octoprint.http

backend letsencrypt-backend
	server letsencrypt 127.0.0.1:54321

backend octoprint
        reqrep ^([^\ :]*)\ /(.*) \1\ /\2
        reqadd X-Scheme:\ https if { ssl_fc }
        option forwardfor except 127.0.0.1
	server octoprint1 127.0.0.1:5000
	acl nautilus hdr_beg(User-Agent) -i nautilus
	acl network_no_password src 192.168.#.#/24
	acl AuthOkay http_auth(L1)
	http-request auth realm octoprint if !AuthOkay !network_no_password !nautilus

backend webcam
        reqadd X-Scheme:\ https if { ssl_fc }
	reqrep ^([^\ :]*)\ /webcam/(.*) \1\ /\2
        server webcam1 127.0.0.1:8080
        server octoprint1 127.0.0.1:5000
	acl network_no_password src 192.168.#.#/24
	acl nautilus hdr_beg(User-Agent) -i nautilus
	acl AuthOkay http_auth(L1)
	http-request auth realm octoprint if !AuthOkay !network_no_password !nautilus

userlist L1
	group G1
	user usera insecure-password ########### groups G1
	user userb password ################# groups G1

Hash your own secure passwords at https://www.mkpasswd.net/index.php?category=crypt and paste them into the userlist section if you don't fancy doing it in an Octopi terminal. (You can also get your users to go and hash their own password and email you the hash if they don't want you to know what their passwords are.)

Happy printing everyone :-)

closed 2017-03-21 by dhardyuk