MoonshineSG/OctoPrint-Mobile #36: Can't get past settings screen

jgdovin, 2017-03-26:
I read the previous issue with a similar problem. I downloaded the app from the apple app store today for my iphone and also installed it through the plugin manager on the octoprint installed on my Raspberry Pi 2.

When I launch the app it says settings need to be set up, please shake to go to settings. I have set up everything in the settings using the API key and url, but when I go back to the app it doesn't leave that screen.

MoonshineSG, 2017-03-26:
Try to "kill" the app and open it again

jgdovin, 2017-03-26:
I have done this, I will try restarting real quick.

MoonshineSG, 2017-03-26:
Killing it should have solve this. Unless the settings have not been "recorded". Check if the api key and the URL are there. 

If doesn't work, last alternative is to remove the app and re download from App Store then, without opening the app, go to the settings and do the setup and only after that open the app. 

I'll do a quick fix for this next week - depending o. Apple approval process  (I'm out on holiday until tomorrow)

jgdovin, 2017-03-26:
I restarted my phone and that fixed it. For whatever reason killing the app did not, maybe killing settings would have. However all is working now, and this is much better than the other app on the store :)

closed 2017-03-26 by jgdovin