MoonshineSG/OctoPrint-Mobile #37: Multiple Printers

S4ndr, 2017-03-26:
Hello, 
i wan't to get this app but is it possible to add multiple printers at once and switch between them?
Thanks!

MoonshineSG, 2017-03-26:
Hi. No, you can only setup one printer. 

I have a "work in progress" version with which I'm trying to  find a way to support multiple printers but it will be a while until it's ready for release. Mostly because it's not really a priority for me - I only have a single printer. But that doesn't mean i am not working on it.

S4ndr, 2017-03-26:
Hello, 
That would be awesome if u can get it! The app looks great and i think i am gonna get it, hope u can figure it out!

stofakiller, 2017-04-04:
I'm also looking for this feature, i have 3 printers with octoprint

MoonshineSG, 2017-04-04:
glad to see that my work will have some use... 
it proves to be slightly more work that I initially anticipated but it's moving forward ... 

btw, it will not be "multi printer monitoring" but "multi printer config" with easy switch between them...

MoonshineSG, 2017-04-27:
the multi printer support is ready. just pending approval from Apple.

closed 2017-04-27 by MoonshineSG

stofakiller, 2017-04-27:
Great, looking forward to try it

MoonshineSG, 2017-04-27:
it took way longer than i expected... (could be because I'm not familiar with objective c ? )