MoonshineSG/OctoPrint-Mobile #38: better way of selecting prints

Austinthemighty, 2017-04-16:
There needs to be a better way of selecting prints other than selecting the last uploaded print.

MoonshineSG, 2017-04-16:
it's supposed to be only a simplified UI for mobile not a full replacement for OctoPrint...

Austinthemighty, 2017-04-17:
I get that but the last uploaded part does not always work and it would be a nice to select what you want to print, also i find the app a lot more reliable than the web interface

MoonshineSG, 2017-04-17:
Reopening this as a reminder that there's interest and in the limit of my time and if I find an easy to implement solution (I hate to think about folders and other functions that might need to be added) I will consider to implement it...

paulobnc, 2017-06-06:
this would definitely be nice. The other octoprint app has it where you can choose the file but has absolutely nothing else which is even worse

MoonshineSG, 2017-06-06:
coming soon... (together with notifications! :D)

1n5aN1aC, 2017-06-06:
NOTIFICATIONS?!

(Incredibly excited for that)

MoonshineSG, 2017-06-06:
yup.. those are fully working now... might release it separately if  proves to be stable enough... 
oh.. and i need to find a unique sound for it :D

MoonshineSG, 2017-06-14:
see release 1.15

closed 2017-06-14 by MoonshineSG

Austinthemighty, 2017-06-23:
Thanks, this has greatly improved the app