MoonshineSG/OctoPrint-Mobile #40: Cannot change IP

NeoTrace82, 2017-04-29:
The IP Adresse of my Printer has been changed, as i did swap my router.
How do i change the IP, cant find anything.

MoonshineSG, 2017-04-29:
which version of the iOS app  do you use ?

NeoTrace82, 2017-04-29:
2.0 did the update yesterday

MoonshineSG, 2017-04-29:
Yes, for 2.0 the settings have moved from the Settings app to the application itself... 
Swipe right from the (screen edge) to get to the screen with all the printers (https://github.com/MoonshineSG/OctoPrint-Mobile/wiki/iOS-App#printer-selection) and then long tap on the printer to access the Edit screen...

NeoTrace82, 2017-04-29:
long tap, did the trick

but now if i change the URL to "http://192.168.0.146:50" it does highlight red if i press Save

MoonshineSG, 2017-04-29:
hmmm... there's a URL check ... the port is the one that fails... 
unfortunately to fix this I need to submit a new app to apple.. gona take a couple of days...

NeoTrace82, 2017-04-29:
ah okay, nevermind
ill have time to wait till you fixed this

thanks for your instant and awesome support :)

MoonshineSG, 2017-04-29:
No worries... just put some stars on the AppStore ;)

MoonshineSG, 2017-04-30:
version 2.0.1 which has the fix for the url check has been submitted to Apple.  Now we wait....

aholesen, 2017-04-30:
I have purchased additional printers but I cannot change my IP for the original printer or add another printer. It fails at the IP adress even though it is correct. Looking forward to 2.0.1 version.

MoonshineSG, 2017-05-01:
@aholesen what is your URL format?

aholesen, 2017-05-01:
http://MYNAME.synology.me:81 and http://MYNAME.synology.me:82

New logo/icon is nice BTW.

MoonshineSG, 2017-05-01:
Ah... it's the port again... sorry about that... I completely forgot testing with port... 
Apple taking their time with the review... 

Thanks! I like the new logo too... :)

MoonshineSG, 2017-05-01:
The fix is out.

aholesen, 2017-05-02:
And it is working fine now - thank you.

closed 2017-05-03 by MoonshineSG