MoonshineSG/OctoPrint-Mobile #45: Switch printer

airiclenz, 2017-05-11:
When having multiple printers set up in the app, there is no way to switch between them. After selecting one printer it try's to connect to that printer in an endless loop.

MoonshineSG, 2017-05-11:
Please explain

airiclenz, 2017-05-12:
I bought the addon for being able to have multiple 3D-printers in Nautilus. I set up 2 printers and used one of them and it works fine. But there seems to be no way to switch to the other printer. The printer I worked with first is turned off. When I then start Nautilus, it tries to load the first printer and I have no chance to switch to the other. Am I missing something? 

I am using an iPhone 7 and an iPad 3. Same problem on both.

1n5aN1aC, 2017-05-12:
It's a bit non-intuitive.

Simply slide in from off-screen left.

MoonshineSG, 2017-05-12:
@airiclenz off-screen slide left to right (please read https://github.com/MoonshineSG/OctoPrint-Mobile/wiki/iOS-App#printer-selection)

@1n5aN1aC when you tap on the printer selection the screen slides right to left... so going back is the reverse... and the sliding off screen is quire common in iOS...  Do you see a better way of doing so ? (the only thing i can think of is to have an arrow popping on the left side to hint that something is there... )

1n5aN1aC, 2017-05-12:
@MoonshineSG Yeah, I did figure it out too, just wasn't immediately obvious.

It may have been more obvious if I was setting it up for the first time, rather than it having already been set up...

closed 2017-05-23 by MoonshineSG