MoonshineSG/OctoPrint-Mobile #46: Connection Issue with app

freakezoid1604, 2017-05-22:
Hello

I downloaded the app from the applestore.
I also install the plugin in octoprint.
When i try to connect i get alway the information "The request timed out."

Thanks for your help

best regars
Freakezoid

MoonshineSG, 2017-05-22:
Using your phone browser (safari or chrome or other) can you connect to the URL you entered in the Nautilus application?

freakezoid1604, 2017-05-23:
Yes I can connect. 
Same link like in the app octo.

freakezoid1604, 2017-05-23:
I try an other thing. 
I add the ip adress like 192.168.1.35, it didn't work then I take http://192.168.1.35 and now it works. 

Strange is that the adress http://octopi.local work in the browser.

MoonshineSG, 2017-05-23:
you mean you added the URL without the "http://" just "192.168.1.35" ? The app doesn't allow you to  save unless the URL starts with `http` or `https`

I'm not sure why ".local" doesn't work either... it should.

Anyways.. now you can connect  ?

freakezoid1604, 2017-05-23:
Yes I'm connected. 
I also changed the rating in the AppStore. 
Maybe other people have the same problem

MoonshineSG, 2017-05-23:
no, there was an old problem with the APIKEY, but that has been long solved.. it's pretty stable now.. thats why I was surprised you had problems... 

Can you explain how did you managed to enter the IP only without `http`. If it's a bug, I want to fix it.. Thanks.

And thank you for removing the bad rating...

closed 2017-05-23 by MoonshineSG