MoonshineSG/OctoPrint-Mobile #47: Request times out

Pappasunrise, 2017-06-10:
The new version doesn't seem to fix my problem unfortunately. I get a very long loading screen followed by "the request timed out" .

MoonshineSG, 2017-06-10:
i see... using the browser on your phone, can you access the same url you configured on the app ?

Pappasunrise, 2017-06-10:
What URL are you referring to?

MoonshineSG, 2017-06-10:
the one you used when you configured your printer in Nautilus.. You had to enter a Name, URL, Api-Key... I'm referring to the URL

Pappasunrise, 2017-06-10:
I downloaded the nautilus plugin on octoprint, I download the app on my phone and when it asks for the API I use the QR code and I select the local, non iCloud option, and then it loads until it fails. Is there something else I need to do first?

Sent with my iPhone

MoonshineSG, 2017-06-10:
yes, you need to configure the URL of your printer... 
how do you access octoprint on your browser ?

Pappasunrise, 2017-06-10:
Just with the octoprint.local http

Sent with my iPhone

MoonshineSG, 2017-06-10:
> I select the local

by that you mean you left it as `http://octoprint.local` ? is that what you type in your browser ?

MoonshineSG, 2017-06-10:
and your iphone is on  the same network/wifi as the printer ?

Pappasunrise, 2017-06-10:
Yes. Same network. 

Sent with my iPhone

MoonshineSG, 2017-06-10:
and if you type `http://octoprint.local` on  the browser on your iphone you can see the octoprint UI ?

Pappasunrise, 2017-06-10:
No I just figured out I need it to be http://octopi.local

Sent with my iPhone

MoonshineSG, 2017-06-10:
oh.... so `octopi` not `octoprint`  ?

Pappasunrise, 2017-06-10:
Yes. 

Sent with my iPhone

MoonshineSG, 2017-06-10:
so now it works ?

Pappasunrise, 2017-06-10:
Mostly. Some of the buttons in the app don't do anything it seems like. I hit the home buttons and nothing happens.

Sent with my iPhone

MoonshineSG, 2017-06-10:
one step at the time...  :D

whats your printer model ? standard gcode ? the code sent by every button can be individually configured in the OctoPrint settings... have a look if your printer is non standard...  turn on the debug and look at the logs.. you can also see the OctoPrint terminal for clues of what it gets sent and why it may not work...

Pappasunrise, 2017-06-10:
The printer I'm using now is a creality CR-10. It uses standard g-code. 

Sent with my iPhone

MoonshineSG, 2017-06-10:
just check the OctoPrint terminal and work out what doesn't work and do the appropriate changes on the settings if needed...

Pappasunrise, 2017-06-10:
It moves when I move it on the app, but it won't home. If I hit the home button from the app the terminal shows nothing. 

Sent with my iPhone

MoonshineSG, 2017-06-10:
check the settings .. it should send G28

Pappasunrise, 2017-06-10:
None of the buttons on the top work

Sent with my iPhone

MoonshineSG, 2017-06-10:
which ones ?

Pappasunrise, 2017-06-10:
All of these

Sent with my iPhone

MoonshineSG, 2017-06-10:
if you're not being specific there is no way for me to help you.

Pappasunrise, 2017-06-10:
I attached a picture. I circled the buttons that don't do anything.

Sent with my iPhone

MoonshineSG, 2017-06-10:
i guess smth went wrong with the attachment... i cant see it...

Pappasunrise, 2017-06-10:
Alright well it's all of the axis homing buttons, as well as the menu button to the right of those

Sent with my iPhone

Pappasunrise, 2017-06-10:
Sent with my iPhone

MoonshineSG, 2017-06-10:
still no picture.. but nevermind...

you can try the following:

- ssh into your RPi... 
- remove the file `/home/pi/.octoprint/data/nautilus/settings.ini` (if will be recreated)
- restart OctoPrint (or reboot the RPi)

and try again...

MoonshineSG, 2017-06-12:
did you manage to get it fixed ?

closed 2017-06-21 by MoonshineSG