MoonshineSG/OctoPrint-Mobile #49: The Request Timed Out

jagvent, 2017-06-28:
I've installed the plugin, installed the iOS app, applied the API key (and make sure it was enabled).

From iOS, I only get "Loading......" then a screen with 'The Request Timed Out'.  

Any ideas?

MoonshineSG, 2017-06-28:
read this carefully and make sure you the highlighted  test works. 
https://github.com/MoonshineSG/OctoPrint-Mobile/wiki/iOS-App#addedit-printer-configuration

jagvent, 2017-06-28:
> wiki/iOS-App#addedit-printer-configuration

Oops.  Was trying to load from the default octoprint.local instead of
octopi.local.  I had to remove the iOS app and reinstall to reconfigure
because couldn't get out of the "Loading" screen.

MoonshineSG, 2017-06-28:
there's a timeout of 2 min... :D 
you can also "kill" the app... but reinstall works as well :D

jagvent, 2017-06-28:
After the time out message, it went back to loading message and looped. When I "killed" the app and restarted, it went right back to loading without opportunity to change the configuration. The only solution was to uninstall.

MoonshineSG, 2017-06-30:
you;re right. it would go back to retry.... swipe right is the only way out...
(not critical, to be fixed in the next release)

closed 2017-06-30 by MoonshineSG