MoonshineSG/OctoPrint-Mobile #51: TypeError: undefined is not a function

bluecamel, 2017-07-25:
I added my printer and now have a full-screen error:

```An error has occured while trying to initialize the plugin. [TypeError: undefined is not a function]```

I can't move away from this screen, even after closing the app and opening again.

MoonshineSG, 2017-07-28:
please  give the version used, the full log from OctoPrint and any other info regarding your config

bluecamel, 2017-07-28:
```
------------------------------------------------------------------------------
OctoPrint version : 1.3.4
OctoPi version    : 0.14.0
------------------------------------------------------------------------------
```

```$ sudo service octoprint restart && sudo tail -f ~/.octoprint/logs/octoprint.log
2017-07-28 16:12:43,419 - octoprint.plugins.announcements - INFO - Loaded channel _important from http://octoprint.org/feeds/important.xml in 0.14s
2017-07-28 16:12:44,084 - octoprint.plugins.announcements - INFO - Loaded channel _releases from http://octoprint.org/feeds/releases.xml in 0.14s
2017-07-28 16:12:44,353 - octoprint.plugins.announcements - INFO - Loaded channel _blog from http://octoprint.org/feeds/octoblog.xml in 0.14s
2017-07-28 16:12:44,617 - octoprint.plugins.announcements - INFO - Loaded channel _plugins from http://plugins.octoprint.org/feed.xml in 0.14s
2017-07-28 16:12:44,931 - octoprint.plugins.announcements - INFO - Loaded channel _octopi from http://octoprint.org/feeds/octopi.xml in 0.14s
2017-07-28 17:04:53,912 - octoprint.server.util.sockjs - INFO - Client connection closed: 192.168.1.155
2017-07-28 19:11:23,483 - octoprint.server - INFO - Shutting down...
2017-07-28 19:11:23,822 - octoprint.events - INFO - Processing shutdown event, this will be our last event
2017-07-28 19:11:23,827 - octoprint.events - INFO - Event loop shut down
2017-07-28 19:11:23,848 - octoprint.server - INFO - Goodbye!
2017-07-28 19:11:27,654 - octoprint.server - INFO - ******************************************************************************
2017-07-28 19:11:27,655 - octoprint.server - INFO - Starting OctoPrint 1.3.4 (master branch)
2017-07-28 19:11:27,656 - octoprint.server - INFO - ******************************************************************************
2017-07-28 19:11:27,772 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2017-07-28 19:11:30,540 - octoprint.plugin.core - INFO - Found 20 plugin(s) providing 20 mixin implementations, 21 hook handlers
2017-07-28 19:11:30,590 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2017-07-28 19:11:31,062 - octoprint.plugin.core - INFO - Found 20 plugin(s) providing 20 mixin implementations, 21 hook handlers
2017-07-28 19:11:31,124 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2017-07-28 19:11:31,176 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2017-07-28 19:11:31,202 - octoprint.plugins.nautilus - INFO - 1 device(s) will receive notifications...
2017-07-28 19:11:31,203 - octoprint.plugins.nautilus - INFO - Nautilus - OctoPrint mobile shell, started.
2017-07-28 19:11:31,225 - octoprint.plugins.multi_colors - INFO - MultiColors init
2017-07-28 19:11:35,365 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python -m pip" as command to invoke pip
2017-07-28 19:11:39,449 - octoprint.util.pip - INFO - Version of pip is 9.0.1
2017-07-28 19:11:39,450 - octoprint.util.pip - INFO - pip installs to /home/pi/oprint/lib/python2.7/site-packages, --user flag needed => no, virtual env => yes
2017-07-28 19:11:39,740 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
2017-07-28 19:11:39,758 - octoprint.plugin.core - INFO - Initialized 20 plugin implementation(s)
2017-07-28 19:11:39,780 - octoprint.plugin.core - INFO - 20 plugin(s) registered with the system:
|  Action Commands (0.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_actioncommands
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/announcements
|  Automatic Shutdown (0.1.4) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_automaticshutdown
|  Autoselect Plugin (0.2.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_autoselect
|  Core Wizard (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/corewizard
|  Cost Plugin (0.1.3) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_cost
|  CuraEngine (<= 15.04) (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/cura
|  Custom Control Editor (0.2.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_customControl
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/discovery
|  DisplayZ (0.1.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_displayz
|  FileManager (0.1.2) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filemanager
|  Fullscreen Plugin (0.0.3) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_fullscreen
|  Gcodebar Plugin (0.1.4) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_GCodeBar
|  Multi Colors (1.0.7) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_multi_colors
|  Nautilus (1.17) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_nautilus
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/pluginmanager
[... cut ...]
2017-07-28 19:11:41,333 - octoprint.plugins.pluginmanager - INFO - Loaded plugin notices data from http://plugins.octoprint.org/notices.json
2017-07-28 19:11:59,365 - octoprint.server.preemptive_cache - INFO - ... done in 18.38s
2017-07-28 19:11:59,367 - octoprint.server.preemptive_cache - INFO - Preemptively caching / (ui _default) for {'query_string': 'l10n=en', 'path': '/', 'base_url': 'http://200.200.200.200/wat/'}
2017-07-28 19:12:06,401 - octoprint.server.preemptive_cache - INFO - ... done in 7.03s```

MoonshineSG, 2017-07-28:
Hmmm... can you try to disable TouchUI ?

bluecamel, 2017-07-29:
I uninstalled TouchUI and rebooted.  No difference.

MoonshineSG, 2017-07-29:
Can you enable debug and post the logs again?

bluecamel, 2017-07-29:
Here's what I just did:
- enabled debugging:
```
loggers:
  octoprint.events:
    level: DEBUG
  octoprint.filemanager:
    level: DEBUG
  octoprint.plugin:
    level: DEBUG
  octoprint.plugin.nautilus:
    level: DEBUG
```
 - uninstalled the nautilus app from my iPhone
 - reinstalled nautilus on my iPhone
 - restarted octoprint
 - got an API key from octoprint
 - added my printer in nautlius
 - tapped on printer in nautilus

Logging settings:

Logs:
```
$ sudo service octoprint restart && tail -f logs/octoprint.log
2017-07-29 14:27:59,446 - octoprint.plugins.announcements - INFO - Loaded channel _releases from http://octoprint.org/feeds/releases.xml in 0.14s
2017-07-29 14:27:59,703 - octoprint.plugins.announcements - INFO - Loaded channel _blog from http://octoprint.org/feeds/octoblog.xml in 0.14s
2017-07-29 14:28:00,037 - octoprint.plugins.announcements - INFO - Loaded channel _plugins from http://plugins.octoprint.org/feed.xml in 0.22s
2017-07-29 14:28:00,382 - octoprint.plugins.announcements - INFO - Loaded channel _octopi from http://octoprint.org/feeds/octopi.xml in 0.14s
2017-07-29 14:28:46,114 - octoprint.server.util.sockjs - INFO - Client connection closed: 192.168.1.155
2017-07-29 16:45:53,295 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.1.155
2017-07-29 16:55:15,435 - octoprint.server - INFO - Shutting down...
2017-07-29 16:55:15,440 - octoprint.events - INFO - Processing shutdown event, this will be our last event
2017-07-29 16:55:15,448 - octoprint.events - INFO - Event loop shut down
2017-07-29 16:55:15,466 - octoprint.server - INFO - Goodbye!
2017-07-29 16:55:19,302 - octoprint.server - INFO - ******************************************************************************
2017-07-29 16:55:19,303 - octoprint.server - INFO - Starting OctoPrint 1.3.4 (master branch)
2017-07-29 16:55:19,304 - octoprint.server - INFO - ******************************************************************************
2017-07-29 16:55:19,419 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2017-07-29 16:55:22,185 - octoprint.plugin.core - DEBUG - Loaded plugin nautilus: Nautilus (1.17)
2017-07-29 16:55:22,187 - octoprint.plugin.core - DEBUG - Loaded plugin actioncommands: Action Commands (0.1)
2017-07-29 16:55:22,188 - octoprint.plugin.core - DEBUG - Loaded plugin virtual_printer: Virtual Printer
2017-07-29 16:55:22,189 - octoprint.plugin.core - DEBUG - Loaded plugin softwareupdate: Software Update
2017-07-29 16:55:22,190 - octoprint.plugin.core - DEBUG - Loaded plugin cost: Cost Plugin (0.1.3)
2017-07-29 16:55:22,192 - octoprint.plugin.core - DEBUG - Loaded plugin automaticshutdown: Automatic Shutdown (0.1.4)
2017-07-29 16:55:22,193 - octoprint.plugin.core - DEBUG - Loaded plugin cura: CuraEngine (<= 15.04)
2017-07-29 16:55:22,194 - octoprint.plugin.core - DEBUG - Loaded plugin displayz: DisplayZ (0.1.0)
2017-07-29 16:55:22,195 - octoprint.plugin.core - DEBUG - Loaded plugin fullscreen: Fullscreen Plugin (0.0.3)
2017-07-29 16:55:22,196 - octoprint.plugin.core - DEBUG - Loaded plugin gcodebar: Gcodebar Plugin (0.1.4)
2017-07-29 16:55:22,197 - octoprint.plugin.core - DEBUG - Loaded plugin slicer: Slicer (1.2.9)
2017-07-29 16:55:22,198 - octoprint.plugin.core - DEBUG - Loaded plugin customControl: Custom Control Editor (0.2.1)
2017-07-29 16:55:22,199 - octoprint.plugin.core - DEBUG - Loaded plugin corewizard: Core Wizard
2017-07-29 16:55:22,200 - octoprint.plugin.core - DEBUG - Loaded plugin autoselect: Autoselect Plugin (0.2.0)
2017-07-29 16:55:22,202 - octoprint.plugin.core - DEBUG - Loaded plugin filemanager: FileManager (0.1.2)
2017-07-29 16:55:22,204 - octoprint.plugin.core - DEBUG - Loaded plugin multi_colors: Multi Colors (1.0.7)
2017-07-29 16:55:22,205 - octoprint.plugin.core - DEBUG - Loaded plugin announcements: Announcement Plugin
2017-07-29 16:55:22,206 - octoprint.plugin.core - DEBUG - Loaded plugin pluginmanager: Plugin Manager
2017-07-29 16:55:22,207 - octoprint.plugin.core - DEBUG - Loaded plugin discovery: Discovery
2017-07-29 16:55:22,209 - octoprint.plugin.core - DEBUG - Enabled plugin nautilus: Nautilus (1.17)
2017-07-29 16:55:22,210 - octoprint.plugin.core - DEBUG - Enabled plugin actioncommands: Action Commands (0.1)
2017-07-29 16:55:22,212 - octoprint.plugin.core - DEBUG - Enabled plugin virtual_printer: Virtual Printer
2017-07-29 16:55:22,213 - octoprint.plugin.core - DEBUG - Enabled plugin softwareupdate: Software Update
2017-07-29 16:55:22,214 - octoprint.plugin.core - DEBUG - Enabled plugin cost: Cost Plugin (0.1.3)
2017-07-29 16:55:22,216 - octoprint.plugin.core - DEBUG - Enabled plugin automaticshutdown: Automatic Shutdown (0.1.4)
2017-07-29 16:55:22,217 - octoprint.plugin.core - DEBUG - Enabled plugin cura: CuraEngine (<= 15.04)
2017-07-29 16:55:22,218 - octoprint.plugin.core - DEBUG - Enabled plugin displayz: DisplayZ (0.1.0)
2017-07-29 16:55:22,219 - octoprint.plugin.core - DEBUG - Enabled plugin fullscreen: Fullscreen Plugin (0.0.3)
2017-07-29 16:55:22,221 - octoprint.plugin.core - DEBUG - Enabled plugin gcodebar: Gcodebar Plugin (0.1.4)
2017-07-29 16:55:22,222 - octoprint.plugin.core - DEBUG - Enabled plugin slicer: Slicer (1.2.9)
2017-07-29 16:55:22,223 - octoprint.plugin.core - DEBUG - Enabled plugin customControl: Custom Control Editor (0.2.1)
[... 315 lines cut ...]
[... cut ...]
2017-07-29 17:04:21,720 - octoprint.events - DEBUG - Subscribed listener <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x6b5b63d0>> for event PrinterProfileDeleted
2017-07-29 17:04:21,721 - octoprint.events - DEBUG - Subscribed listener <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x6b5b63d0>> for event CaptureDone
2017-07-29 17:06:07,495 - octoprint.plugins.nautilus - INFO - Registering device with token [a1b4cae47bcc0a7b81f8950f384d307420d9e272e34e812f8474e664a0d6d952]
```

bluecamel, 2017-07-29:
I should mention that I do get notifications through the app when a print finishes or is canceled, so it is talking to octoprint.  But I can't do anything else in the app after going to the printer.  I can't even go back to the menu to change printer settings.

bluecamel, 2017-07-29:
Ah, I see that I didn't enable debugging correctly for nautilus.  Let me do all of that again.

MoonshineSG, 2017-07-29:
To get back to the menu swipe right from the edge of the screen

MoonshineSG, 2017-07-29:
iOS version ? Device?

bluecamel, 2017-07-29:
Alright, changed `logging.yaml` to:
```
loggers:
  octoprint.events:
    level: DEBUG
  octoprint.filemanager:
    level: DEBUG
  octoprint.plugin:
    level: DEBUG
  octoprint.plugins.nautilus:
    level: DEBUG
```

The logs still show that nautilus is logging at `INFO` level:
```
$ sudo service octoprint restart && tail -f logs/octoprint.log
2017-07-29 17:11:27,029 - octoprint.events - DEBUG - Unsubscribed listener <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x6af7e9d0>> for event PositionUpdate
2017-07-29 17:11:27,029 - octoprint.events - DEBUG - Unsubscribed listener <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x6af7e9d0>> for event SlicingStarted
2017-07-29 17:11:27,030 - octoprint.events - DEBUG - Unsubscribed listener <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x6af7e9d0>> for event PrinterProfileDeleted
2017-07-29 17:11:27,031 - octoprint.events - DEBUG - Unsubscribed listener <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x6af7e9d0>> for event CaptureDone
2017-07-29 17:17:17,150 - octoprint.server - INFO - Shutting down...
2017-07-29 17:17:17,204 - octoprint.events - INFO - Processing shutdown event, this will be our last event
2017-07-29 17:17:17,206 - octoprint.events - DEBUG - Firing event: Shutdown (Payload: None)
2017-07-29 17:17:17,208 - octoprint.events - DEBUG - Sending action to <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x6b5b63d0>>
2017-07-29 17:17:17,215 - octoprint.events - INFO - Event loop shut down
2017-07-29 17:17:17,231 - octoprint.server - INFO - Goodbye!
2017-07-29 17:17:21,032 - octoprint.server - INFO - ******************************************************************************
2017-07-29 17:17:21,033 - octoprint.server - INFO - Starting OctoPrint 1.3.4 (master branch)
2017-07-29 17:17:21,034 - octoprint.server - INFO - ******************************************************************************
2017-07-29 17:17:21,148 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2017-07-29 17:17:23,843 - octoprint.plugin.core - DEBUG - Loaded plugin nautilus: Nautilus (1.17)
2017-07-29 17:17:23,845 - octoprint.plugin.core - DEBUG - Loaded plugin actioncommands: Action Commands (0.1)
2017-07-29 17:17:23,846 - octoprint.plugin.core - DEBUG - Loaded plugin virtual_printer: Virtual Printer
2017-07-29 17:17:23,848 - octoprint.plugin.core - DEBUG - Loaded plugin softwareupdate: Software Update
2017-07-29 17:17:23,849 - octoprint.plugin.core - DEBUG - Loaded plugin cost: Cost Plugin (0.1.3)
2017-07-29 17:17:23,850 - octoprint.plugin.core - DEBUG - Loaded plugin automaticshutdown: Automatic Shutdown (0.1.4)
2017-07-29 17:17:23,851 - octoprint.plugin.core - DEBUG - Loaded plugin cura: CuraEngine (<= 15.04)
2017-07-29 17:17:23,852 - octoprint.plugin.core - DEBUG - Loaded plugin displayz: DisplayZ (0.1.0)
2017-07-29 17:17:23,853 - octoprint.plugin.core - DEBUG - Loaded plugin fullscreen: Fullscreen Plugin (0.0.3)
2017-07-29 17:17:23,854 - octoprint.plugin.core - DEBUG - Loaded plugin gcodebar: Gcodebar Plugin (0.1.4)
2017-07-29 17:17:23,855 - octoprint.plugin.core - DEBUG - Loaded plugin slicer: Slicer (1.2.9)
2017-07-29 17:17:23,856 - octoprint.plugin.core - DEBUG - Loaded plugin customControl: Custom Control Editor (0.2.1)
2017-07-29 17:17:23,857 - octoprint.plugin.core - DEBUG - Loaded plugin corewizard: Core Wizard
2017-07-29 17:17:23,859 - octoprint.plugin.core - DEBUG - Loaded plugin autoselect: Autoselect Plugin (0.2.0)
2017-07-29 17:17:23,861 - octoprint.plugin.core - DEBUG - Loaded plugin filemanager: FileManager (0.1.2)
2017-07-29 17:17:23,862 - octoprint.plugin.core - DEBUG - Loaded plugin multi_colors: Multi Colors (1.0.7)
2017-07-29 17:17:23,864 - octoprint.plugin.core - DEBUG - Loaded plugin announcements: Announcement Plugin
2017-07-29 17:17:23,865 - octoprint.plugin.core - DEBUG - Loaded plugin pluginmanager: Plugin Manager
2017-07-29 17:17:23,866 - octoprint.plugin.core - DEBUG - Loaded plugin discovery: Discovery
2017-07-29 17:17:23,868 - octoprint.plugin.core - DEBUG - Enabled plugin nautilus: Nautilus (1.17)
2017-07-29 17:17:23,870 - octoprint.plugin.core - DEBUG - Enabled plugin actioncommands: Action Commands (0.1)
2017-07-29 17:17:23,871 - octoprint.plugin.core - DEBUG - Enabled plugin virtual_printer: Virtual Printer
2017-07-29 17:17:23,873 - octoprint.plugin.core - DEBUG - Enabled plugin softwareupdate: Software Update
2017-07-29 17:17:23,874 - octoprint.plugin.core - DEBUG - Enabled plugin cost: Cost Plugin (0.1.3)
2017-07-29 17:17:23,875 - octoprint.plugin.core - DEBUG - Enabled plugin automaticshutdown: Automatic Shutdown (0.1.4)
2017-07-29 17:17:23,877 - octoprint.plugin.core - DEBUG - Enabled plugin cura: CuraEngine (<= 15.04)
2017-07-29 17:17:23,878 - octoprint.plugin.core - DEBUG - Enabled plugin displayz: DisplayZ (0.1.0)
2017-07-29 17:17:23,880 - octoprint.plugin.core - DEBUG - Enabled plugin fullscreen: Fullscreen Plugin (0.0.3)
2017-07-29 17:17:23,881 - octoprint.plugin.core - DEBUG - Enabled plugin gcodebar: Gcodebar Plugin (0.1.4)
[... cut ...] leted
2017-07-29 17:18:08,617 - octoprint.events - DEBUG - Subscribed listener <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x6b633b70>> for event CaptureDone
2017-07-29 17:18:10,365 - tornado.access - WARNING - 401 POST /api/plugin/automaticshutdown (127.0.0.1) 43.57ms
2017-07-29 17:19:07,079 - octoprint.plugins.nautilus - INFO - Registering device with token [882ea6fa229c433173424f6a54b2523172ff9aa33e54116914c24306e353ca10]
```

MoonshineSG, 2017-07-29:
The error seams to be on the client side and it happens then the iOS app calls the javascript which somehow is not loaded or partial loaded... Hard to debug... 
The logs from the static files would be good. Let me get back and check what else I would need in order to help. Early next week.

bluecamel, 2017-07-29:
Sorry, I missed your responses between my posts.  iOS version is 10.3.2.  Device is iPhone 6 Plus.  I see that I can upgrade to 10.3.3, so I'll try that too.  I can also try my iPad.

Do you have any advice on how to get debug mode turned on for nautilus?  Also, you mention logs from static files.  How do I get those?

If there's anything else I can do to debug, please let me know.

bluecamel, 2017-07-29:
I figured it out thanks to your comments!  I have nginx as a reverse proxy and I didn't have octoprint configured correctly to use the prefix, scheme, and host headers.  Instead, I had a `prefixFallback`, which was causing the problem.

Sorry to bug you when it was completely my fault, but thanks so much for the help!  The app seems great so far!

closed 2017-07-29 by bluecamel

MoonshineSG, 2017-07-29:
no worries, just great that you have it working!!!

bluecamel, 2017-07-29:
Yeah, I'm excited to use it!  I can already tell that I'm going to love it for changing filament and doing cold pulls.  That's always a pain through the other UIs, but the temp presets will make it a breeze!

MoonshineSG, 2017-07-29:
Yup... those are few of the reasons why I build it. For my own use because anything was just not friendly enough... 
you might also like the ability to change  the gcode commands for each button...

A positive feedback on iTunes and spreading the word would be appreciate if you enjoy the app.

bluecamel, 2017-07-29:
Thanks, I'm looking into all of the settings now that I have things working.  I'm also planning to setup the switches for LEDs and power control (but still working on the hardware side there).  Just as exciting, if I'm interpreting this plugin correctly, I can just edit the templates, js, etc., and the app will load those.  If so, I will literally dance since I can easily make a web app to do my bidding without touching iOS code!

MoonshineSG, 2017-07-29:
Yup... it’s a  webapp and most of it can be easily modified. Authentication part should remain and if somehow in the future that changes, you might have to manually patch the codes... 
but otherwise, free to do your own thing... the iOS app is just a shell that load the webapp and maintains configs and shows notifications ...

MoonshineSG, 2017-07-29:
Oh... and shake the device to reload the app ;)