MoonshineSG/OctoPrint-Mobile #52: Video stream not working

dvilo, 2017-07-27:
I am using IP webcam and am able to view stream on iPhone from Octoprint/control and directly from IP webcam phone.

From within Nautilus I sometimes immediately get "Sorry, camera is not available.", other times it tries to load for about 20 seconds before the message is displayed. I have never been able to load the stream, even after reboots of both phone and pi.

Octoprint stream URL is set to http://192.168.1.xxx:8080/video (IP Webcam default streaming URL).

Any ideas on how to get the stream working?

MoonshineSG, 2017-07-27:
The video stream url is taken from octoprint settings .

dvilo, 2017-07-27:
Is any formatting done to the URL? Since stream is working in Octoprint it seems odd that it does not load in Nautilus.

MoonshineSG, 2017-07-27:
No formatting is done. Simply taken from the settings. Can you access it on the octoprint on the iPhone?

dvilo, 2017-07-27:
Yes, octoprint on iPhone is ok.

MoonshineSG, 2017-07-27:
Strange. 
I am on holiday now but i’ll check next week when I get back home. 
Meanwhile can you have a look at the logs and see if you can spot any error messages?

dvilo, 2017-07-27:
Further info: stream ratio is 4:3, could this have any unwanted effects?

dvilo, 2017-07-27:
Thanks. I will keep you posted on my findings.

MoonshineSG, 2017-07-27:
Ratio would give you a distorted image in the worst scenario...

MoonshineSG, 2017-07-28:
is the url properly setup in the OctoPrint ?

MoonshineSG, 2017-08-07:
did you managed to get this working ?

closed 2017-08-08 by MoonshineSG