MoonshineSG/OctoPrint-Mobile #54: Self signed SSL?

gitterdoneplease, 2017-08-06:
I get a message about certificate not valid, etc., but cannot bypass. Is this a supported config?

MoonshineSG, 2017-08-07:
See https://github.com/MoonshineSG/OctoPrint-Mobile/wiki/SSL-Error

MoonshineSG, 2017-08-07:
I suggest you use a free "proper" SSL cert if continues to be a problem for you.

closed 2017-08-08 by MoonshineSG

RaphPi, 2017-08-29:
Sorry but it doesn't work for me. I've made a new certificat and do the process ssl-error on iPhone but always the same request "the certificate for this server is invalid".

MoonshineSG, 2017-08-29:
did you install the created cert on the iphone ?

RaphPi, 2017-08-29:
Yes, it is installed and trusted.

RaphPi, 2017-08-29:
It is a .crt file. I've followed the suggested link bellow.

MoonshineSG, 2017-08-29:
is the certificate issued correctly, date, domain , etc  ?  and installed on the haproxy ? can you access - without error - the secured url from a browser -safari or chrome - on the iphone  ?

gitterdoneplease, 2017-08-29:
@RaphPi I had the precise same issue. The only way to make it work is to create your own root, sign the cert and upload the root cert to phone and trust that.

RaphPi, 2017-08-29:
@MoonshineSG on iPhone i can access to octoprint.

@gitterdoneplease That's what i do. Maybe i've made a mistake. What is the exact process for you ?

I've followed this tutorial to generate : https://github.com/PrusaMK2Users/MK2_Tips_and_Tricks/wiki/Generating-a-new-SSL-certificate-for-OctoPi

RaphPi, 2017-08-29:
The strange things is that i receive notifications...

gitterdoneplease, 2017-08-29:
A bit more complex. Something like this.

https://datacenteroverlords.com/2012/03/01/creating-your-own-ssl-certificate-authority/

MoonshineSG, 2017-08-30:
When you say that you can access octoprint, is that using https ? 
I have tested with self issued cert and with own root as well as proper ssl certificate. Once the iPhone recognizes the cert as trusted the Nautilus app does the same.

Alternatively you can get a proper ios recognized certificate. The ones from Let's Encrypt work great and are free.

Notifications are sent to the ios device by the apple servers and is independent from the way you access your printer.

gitterdoneplease, 2017-08-30:
Let's Encrypt needs you to renew every 90 days. You run that on your Pi?

MoonshineSG, 2017-08-30:
Yes, 90 days. But the renewal can be automated via a cron job. 
Since I found out about it, I replaced all my self issued certs with it and works like a charm!!

gitterdoneplease, 2017-08-30:
Fair play sir. I'll look into it further. Seemed like a PITA.

MoonshineSG, 2017-08-30:
once setup, you can (and will) forget about it... 

btw, if you only access the printer locally, you don't need https. 

If you access it from the internet, depending how you access it, the SSL cert would be needed on the proxy, if you use one.
If you just to a port redirect, then yes, the SSL cert should be on the RPi.

RaphPi, 2017-08-30:
@MoonshineSG and @gitterdoneplease Thanks for your help but I'm stuck.
I've followed the tutorial and put the root and the device certificate on iPhone (See the device certificate on iPhone look good) but Nautilus don't want it.

Maybe you have the exactly command lignes I need to use. Is there something to change on haproxy.cfg ?

[image]

MoonshineSG, 2017-08-30:
the certificate you installed in ios is the root or the cert itself ? 

in my case (i use Let's Encrypt) I merged the `certificate.key` + `certificate.crt`  + `intermediate.crt ` into a single `certificate.pem` file and then the haproxy config has 

```
frontend public
        bind *:80
        bind 0.0.0.0:443 ssl crt /etc/ssl/certificate.pem

```

RaphPi, 2017-08-30:
I install the root and the cert. If I put the cert without the root the cert is not trusted.
On the server I merge cert.key and cert.crt in snakeoil.pem. so I normally not need to changed the haproxy config file.

`frontend public
bind *:80
bind 0.0.0.0:443 ssl crt /etc/ssl/snakeoil.pem`

MoonshineSG, 2017-08-30:
Sounds correct... not sure why is there an issue

RaphPi, 2017-08-30:
With http connection it works. So it's really with the certificate.
Maybe the informations in the certificate are false. I don't know exactly what to put on domain. I try with the Internet IP+port or a name. it works on safari for the both. So what verifications do the apps that safari doesn't realize?

MoonshineSG, 2017-08-30:
it's usually domain name but IP (no port) should work. the port is 443 almost alway ..