MoonshineSG/OctoPrint-Mobile #55: App opens, but no printer interface is loaded

caesarius, 2017-08-31:
The app is loaded, but it freezes at the Loading page.

MoonshineSG, 2017-08-31:
Please give more details about your setup. There is no way to know what could be wrong by saying "it freezes"

caesarius, 2017-08-31:
The initial configuration is done, then it just freezes at the Loading screen… I'd love to give you more details, but I don't have anything else.
iPhone 6S MKQN2LZ/A, iOS 10.3.3 (14G60)

caesarius, 2017-08-31:
Also, OctoPi attached to a Original Prusa MK2S.

MoonshineSG, 2017-08-31:
http or https ? 
APIKEY setup ? 
the URL entered in the config page works on safari on the iphone ? (shoudl show OctoPrint UI)

MoonshineSG, 2017-08-31:
nautilus plugin installed ?
errors in the octoprint logs ?

caesarius, 2017-08-31:
- HTTP
- API Key setup
- It works on mi computer browser and iPhone browser
- Nautilus (1.17) plugin installed
- No errors

BUT nevermind, I was entering http://octoprint.local (the default URL on the configuration screen), instead http://octopi.local (my current setup).

So maybe there should be a way to view/edit the configuration instead launching it automatically. I uninstalled and reinstalled the app in order to do that.

MoonshineSG, 2017-08-31:
there's no need to uninstall...  you can edit the configuration. Just swipe left to right from the left side of the screen (in landscape mode)
 
see https://github.com/MoonshineSG/OctoPrint-Mobile/wiki/iOS-App#addedit-printer-configuration

caesarius, 2017-08-31:
Thanks for the clarification. I would suggest making it explicit, having to read the manual for a core functionality is not ideal. Maybe a series of onboarding screens the first time the app is launched? My 5 cents.

closed 2017-08-31 by caesarius