MoonshineSG/OctoPrint-Mobile #56: constant loading screen

race12001, 2017-09-06:
I installed the plugin on my iPhone 7 and my Pi but all it says is loading. I did get it to day timed out once but I can not get past the loading screen on the phone to set it up.

MoonshineSG, 2017-09-07:
have a look at https://github.com/MoonshineSG/OctoPrint-Mobile/issues/55 and check out the "solution"

MoonshineSG, 2017-09-17:
did you manage to get this working ?

race12001, 2017-09-17:
I have not. I also have not had much time at the moment to test it, I've been moving / remodeling a house.

MoonshineSG, 2017-09-18:
ok. anyway, I have a new version coming out for iOS 11 that handles better the timeouts and errors.

MoonshineSG, 2017-10-03:
the new version is been out for a while. hopefully it solved your problem. let me know when you have time

closed 2017-11-21 by MoonshineSG