MoonshineSG/OctoPrint-Mobile #58: iCloud sync not working

cpmorris, 2017-09-26:
Have the app on several devices but settings and purchases are not syncing properly.  iCloud it enabled for the app

MoonshineSG, 2017-09-26:
I assume the iCloud is enebeled using the same account?
When you said “not synced properly” what do you mean? Not synced at all or some data gets mixed up? (That would be odd!)

cpmorris, 2017-09-26:
Yes all the same account and not synced at all.

MoonshineSG, 2017-09-26:
The first iCloud sync is sometimes unpredictable... 
Reinstall the app (one one of the devices) and if it gets synced try reinstalling one by one ... 
once a first sync is done the updates should work without problem

MoonshineSG, 2017-09-26:
Or even switch iCloud off then on (per app) might help

cpmorris, 2017-09-30:
Tried all that, still no luck.

Any other ideas??

MoonshineSG, 2017-10-01:
So, you have two devices. One is fully setup and working with one or more printers. The second one was ok but after a re-install it does not sync with iCloud. Did I get that right ? 

If that is the case I suggest the following steps (to force iCloud sync)
- delete the app on the device that is NOT working (#2)
- make sure the one that is working (#1) has iCloud enabled. 
- slightly modify one of the printer configs on #1 (change the name -you can change it back later)
- reboot the non working device #2 (maybe both?)
- install the app on the second device #2
- open it once and wait a short while.
- reboot #2 
- open the app on #2 and wait. 
- if the data doesn't show, simply close the app and have a cup of coffee or a beer, then try again...

Hopefully this will work. I only had problems with iCloud not showing a few times but ALWAYS it does eventually sync.

cpmorris, 2017-10-01:
The second device has never sync’d but first one has two connections now. I’ll try the list below and see what happens.

Thanks

rstcologne, 2017-10-06:
Hi,

I have the same issue. 3 devices, fresh installation a couple of days ago. I set up a printer on one device and it just won't sync to the other devices. iCloud is enabled on all three and works. I have other apps that sync via iCloud and they sync just fine. 

I tried the steps you mentioned above but it just doesn't sync. Not even after several hours. 

All my devices are on iOS 11.0.1 at the moment.

MoonshineSG, 2017-10-06:
I am really at lost here. I have many times removed the app on mine of my devices just to test how the iCloud sync works and even if though the data will always make it across the timing is in no way consistent. Sometimes it happens in seconds (rarely) or minutes, but once in a while i just give up for a while, but when I try again in X min (hours?) the data has been synced. 

I wish I could have more control, but there is no "force sync" that the app can trigger. 

Note (if that was not clear): iCloud has to be using the same account for the data to be synced.

rstcologne, 2017-10-06:
I just tried again after the 10.0.2 iOS update. Device reset, app reinstall on all the devices. No luck. And yes, all my devices use the single apple/icloud account I have. Still no luck, not even after hours of waiting. 

Is there any way I can support debugging this?

rstcologne, 2017-10-06:
@cpmorris Syncing of purchases works for me via the restore purchases mechanism but has to be done manually. I suppose that's similar to many other apps out there.

MoonshineSG, 2017-10-06:
Restore purchases is a great idea. So that triggered the iCloud sync for you?

cpmorris, 2017-10-06:
It wasn’t working on my other devices before but I’m restored and syncing now
Thanks for all the help

rstcologne, 2017-11-21:
It still doesn't sync, even though you closed the bug report. Since my initial comments, quite a few device reboots, IOS upgrades took place, I reinstalled the app a number of times. No change. I have a few applications which use icloud sync and they all work with my devices. It's not a blocker since an easy workaround is to just add all printers in all devices manually, however this issue still exists at least for some of your users/customers.

MoonshineSG, 2017-11-21:
I’ll keep it open if it still happens to you...

rstcologne, 2017-11-21:
Is there anything I can to to help debug this?

MoonshineSG, 2017-11-21:
no, not really.
While developing this, it happened to me a few times but it kinda self corrects after a while. Apple doesn't allow apps to trigger a sync. but instead apps receive a notification when a sync occurs. 
I can have a "debug" version of the app but there isn't much it can show. 
Let me do some digging and see if theres smth I missed and then maybe I'll ask you to install another (test) version

MoonshineSG, 2017-11-21:
restoring the purchase doesn't work for you ?

rstcologne, 2017-11-21:
No, it does restore the purchases but still no syncing.

MoonshineSG, 2017-11-21:
I wonder what’s so special about your devices...

rstcologne, 2017-11-21:
me, too :-) I wasn't aware that there was anything special about them. One of them is MDM controlled however the other two are just normally used.

MoonshineSG, 2017-11-21:
MDM?
And all logged in the same iCloud account... if you edit a printer on one device (change name for ex) that change will not show on all 3 devices?

rstcologne, 2017-11-21:
MDM = Mobile Device Management. Yes, all devices are logged in with the same iCloud account. I only have one. I have other apps that use icloud to sync/access icloud and they work/see the same data on icloud.

My printer's AirPrint/Bonjour name needs to be set on the printer. Can I even change it on the device?

MoonshineSG, 2017-11-21:
The name of the printer as setup in Nautilus. Touch and hold to go into edit mode

rstcologne, 2017-11-21:
Ah, sorry, I was on the wrong track there... No, these kinds of changes do not replicate. I can add/remove/modify printers on either of the devices and these changes will not show up on the others. I can delete the app on a device, reinstall and none of the printer setups show up on the reinstalled device, even after days and restoring purchases. All devices act as if there was no syncing functionality. They all have and keep their own individual setups as manually created on each of the devices.

MoonshineSG, 2017-11-21:
happens regardless on which device you make the changes ?

rstcologne, 2017-11-21:
Yes, I tried all sorts of scenarios. Start on device A, set up a printer, Install app on Device B, wait. Do it the other way round. Make changes on A or B. That was with my personal iPad Air and iPhone 6s plus. I then also added my company iPhone to the mix (the one that's under MDM). I tried your recipie from above with the reboots in both directions A to B as well as B to A. It just doesn't sync anything.

MoonshineSG, 2017-11-21:
can I declare you a lost case ?
I really don't know what else to suggest...

If I have some time, I'll do a "special" app and ask you to test if you don't mind... it really bugs me that I don't know why this happens... 

I can only hope you're a unique case. so far nobody complained...

rstcologne, 2017-11-21:
Sure. Happy to do the test. 

Should I be the only one, I guess lost cause might be an option. I can live with it. I only have 2 printers, setting them up individually is not that much trouble. Just thought you might want to know that there's at least one user where sync doesn't work.

MoonshineSG, 2017-11-24:
On one or all your devices you have disabled iCloud for the Nautilus app. Once you enable that, the sync should work.
I have a new version(pending Apple review) that will display and error if iCloud is not available. 
You can wait for that and you’ll see which device doesn’t have access. 
Pretty sure this will fix your issue.

closed 2018-01-01 by MoonshineSG

overling, 2019-10-13:
icloud not available.  used to work on this phone. app only in this phone.

MoonshineSG, 2019-10-14:
> icloud not available. used to work on this phone. app only in this phone.

is your iCloud enabled for Nautilus ? See: Settings -> [Apple ID] - iCloud - Apps Using iCloud - Nautilus

[image]

overling, 2019-10-14:
yes its enabled.  The one thing that has changed on my end is phones were updated from IOS 12 to 13.

MoonshineSG, 2019-10-14:
So when you say “iCloud not available “ what exactly do you refer to? Only having iCloud access turned off would show that error message .... Is that the message you see on the main screen? 

 (btw, nautilus under iOS 13 works fine )

overling, 2019-10-14:
The blue screen where it shows the different profiles that can be logged into.  I used to have 2 or 3, one for home, one for outside the house and there was a + button.  Now there is just one profile and where the + normally is it says something about icloud not available.   I did look in the phone settings and icloud is enabled for the nautilus app.  Im not with the phone at the moment to take a screen shot.  Is there any other places besides that icloud app toggle switch that need to be enabled?  I did attempt to remove and reinstall the app but same issue.  The paid profiles are not showing up just the main one.,

MoonshineSG, 2019-10-14:
No other place to change iCloud access... I was about to ask you to remove the app, restart the phone and the reinstall it... 
do you have iCloud issues on any other app?

MoonshineSG, 2019-10-14:
another thing you could try is to logout icloud and login again...

overling, 2019-10-14:
[image]

This is the only app I am aware of with an issue. I will try logging out and back in after work.  Thanks.