MoonshineSG/OctoPrint-Mobile #59: Name resolution not working

hutchyx, 2017-10-03:
DNS Name resolution not working beyond the first 3 printers.   When setting up the fourth printer and beyond, it won't resolve the dns name, it'll only work via IP

MoonshineSG, 2017-10-03:
DNS resolution is done by your network. Maybe smth wrong with the server name itself ?

hutchyx, 2017-10-03:
I do get name resolution from other apps using the same name.  It's only started happening after ios11 update.

MoonshineSG, 2017-10-03:
could be a typo error ?

MoonshineSG, 2017-10-03:
what is the exact error message ?

hutchyx, 2017-10-03:
Definitely not a typo.  I can copy and paste the same name to another app and get name resolution.  
It just flashes the back color of the octoprint url box to pink.   
I was incorrect earlier in saying it only happens beyond the first 3.   It seems to happen if I edit any of the existing profiles or try to add a new one.   
The existing profiles, unedited, work ok using names.

MoonshineSG, 2017-10-03:
Ah... thats not the DNS... the app does a regex check to ensure a valid URL. 
If it fails for you, this is smth I need to fix. Can you tell me the url ? (change some words - for privacy -but keep the format - numbers, special characters, etc...)

MoonshineSG, 2017-10-03:
and.. what version of the app are you using ?

hutchyx, 2017-10-03:
Hyphens

hutchyx, 2017-10-03:
Is there a way to see the version number? I have the most current from the iOS app store for iPhone

MoonshineSG, 2017-10-03:
yup... tested on the latest version...failed. I'll have a fix and submit to apple... give me a few days. sorry.

hutchyx, 2017-10-03:
Thanks! 
I thought I was losing my mind

MoonshineSG, 2017-10-03:
version can be found in the iOS Settings / Nautilus

MoonshineSG, 2017-10-03:
don't worry, you're sane!

hutchyx, 2017-10-03:
2.1.2

MoonshineSG, 2017-10-04:
New app is watch review. I left very basic checks so most (even rubbish) URL should pass. It will take a day or so for Apple to review the app.

MoonshineSG, 2017-10-05:
New version is out. Update the app and try again.

hutchyx, 2017-10-05:
Resolved
Thank you!

closed 2017-10-05 by MoonshineSG

MoonshineSG, 2017-10-05:
@hutchyx: positive (if justified) review/rating on AppStore is alway welcomed :D