MoonshineSG/OctoPrint-Mobile #60: connectiosn server status

cpmorris, 2017-10-19:
Has been working fine for weeks now my first install wont connect to the server and I cant change the config file.  I'm still able to control from the app but cant customize functions.  I tried uninstalling the plug in and reinstalling but that didn't fix it.  My other printer is working fine.

THanks

MoonshineSG, 2017-10-20:
Sorry, i’m not sure what you mean

cpmorris, 2017-10-20:
[image1.jpeg]

MoonshineSG, 2017-10-20:
Try attaching the image again

cpmorris, 2017-10-20:
[image1.jpeg]

MoonshineSG, 2017-10-20:
Nope

cpmorris, 2017-10-20:
When I click on nautilus under plug ins it says at the top: “notification server status:connection problem?”  And the config area is blank.  It had been working fine.  I’ve completely remove and installed the plug in and rebooted.

MoonshineSG, 2017-10-20:
What version octoprint? I am away on holiday and haven’t check the plugin on the latest update... maybe smth is broken... 
can you see anything on the logs?
Your printer is connected to internet ?

MoonshineSG, 2017-10-20:
How about when you use the iPhone app? Any problem? Or is just the settings page on octoprint has issues?

cpmorris, 2017-10-20:
Both of my setups are running 1.3.5

MoonshineSG, 2017-10-20:
One works and one doesn’t?

cpmorris, 2017-10-20:
Correct. Everything from the app works I just can’t customize the commands on the one that’s got the issue.

MoonshineSG, 2017-10-20:
Strange. Both can connect to the internet ? 
Can you get the logs on the one that doesn’t work?
Any additional plugins running ?

cpmorris, 2017-10-20:
Yes both have internet and same additional plug in. I’ll try to get the logs later this morning

MoonshineSG, 2017-11-21:
managed to get it working ?

cpmorris, 2017-11-21:
Yes it is back to normal now.  Didn't do anything specific.

MoonshineSG, 2017-11-24:
maybe it was just some network issue. I'll close this for now, but feel free to reopen if you face more problems.

closed 2017-11-24 by MoonshineSG