MoonshineSG/OctoPrint-Mobile #61: Webcam Display in iOS App

digitaljunk, 2017-11-19:
Cant get an network camera to work in the App.
Suggestions? 
iOS 11.1.2
OctoPrint 1.3.5
Nautilus 1.18
Stream URL: http://IP/video.cgi
on LAN/WIFI

MoonshineSG, 2017-11-20:
Is it setup in OctoPrint ?
does it work in OctoPrint ? 
the `IP` is the same as the OctoPrint IP ?

digitaljunk, 2017-11-21:
Yes, its in the octoprint settings.
Its running fine in octoprint. 
No, cam has different IP. The IP is on the same LAN and IP range.

MoonshineSG, 2017-11-21:
so, the OctoPrint settings has `http://IP1/video.cgi`  setup and works well when accessed on a browser on `http://IP2`, but not on Nautilus ? Nautilus is setup to access  `http://IP2`. Is this right ?
 
Does the rest of UI works on Nautilus ? Any error message ("cam not available"?) or just an empty screen ?

digitaljunk, 2017-11-21:
Thats correct. IPs are different.
"Sorry, Camera is not available." Error Message.

MoonshineSG, 2017-11-21:
can you open `http://IP/video.cgi` from a **browser** on the iPhone ?

digitaljunk, 2017-11-21:
Yes, it works in Safari.

MoonshineSG, 2017-11-21:
thats normal, as it is streaming and it is constantly loading...

are you able to SSH to the machine where OctoPrint/Nautilkus is installed ?

digitaljunk, 2017-11-21:
Yepp, but i need some sleep now. :( I am following this tomorrow again. :)

MoonshineSG, 2017-11-21:
never mind the SSH... there will be a fix by the time you wake up.
update the plugins and it should be ok

digitaljunk, 2017-11-22:
its fixed! Thank you for your awesome Support.

MoonshineSG, 2017-11-24:
Youțre welcome. A positive rating/review on App Store would be much appreciated :D

closed 2017-11-24 by MoonshineSG