MoonshineSG/OctoPrint-Mobile #62: iPhone X Display issues

trendelkamp, 2018-01-06:
Hi,

on an iPhoneX the Display is not utilized 100% correct. 
In menus icons are cut a little bit on the sides. 
More of an issue is the scaling of the webcam stream. This is streched. 
Please see the attached screenshots. 

Thanks
[image]
[image]
[image]

MoonshineSG, 2018-01-07:
in order to maximise the usable UI, the icons have to suffer a little (so it's on purpose like that!) ... nothing really that affects usability... when apple removes that notch, you will be able to see the full icons...

as for the camera, it is "fit" mode on height - so you can see whole image, but if you tap on it it changes to "full screen" and follows the camera proportions and allows pinch to zoom... (tap again to go back to the menu mode)

MoonshineSG, 2018-01-08:
I did some minor changes to deal with some UI issues under X... update to 1.20 and have a look

trendelkamp, 2018-01-11:
Looks superb after the update! Thanks
For the camera, didn´t know tapping on the screen hasn an effect with the camera... ;)

Case closed

closed 2018-01-11 by trendelkamp

MoonshineSG, 2018-01-11:
Btw, if you enjoy Nautilus, a positive review on AppStore would be much appreciated. Thanks.