MoonshineSG/OctoPrint-Mobile #63: Connect over IP through Haproxy?

devsfan1830, 2018-01-12:
I have Haproxy installed in order to secure my printer from unwanted remote access. So my IP access requires a different port and a password load up the octoprint interface. Can Nautilus be setup to authenicate with Haproxy?

MoonshineSG, 2018-01-12:
Do you refer to http authorisation? That is not currently supported. 
you could add a rule based on user agent ("nautilus") and then the apikey would be the one protecting the url

MoonshineSG, 2018-01-13:
No. If I understand correctly, you added "basic access authentication" to your haproxy for the OctoPrint backend.
What I am suggesting is to allow requests coming from nautilus - selected based on http header "User Agent" - to bypass the need of authentication.

Hope this makes it clearer. 
I am also looking how difficult would be to support basic access authentication in Nautilus... But that might take some time...

MoonshineSG, 2018-01-13:
just curious... have you tried to put the URL as "https://user:password@octoprint.local" ?

devsfan1830, 2018-01-13:
octoprint.local doesn't require any authentication as it only works when I'm home on my own network. Frankly I'm not the one who setup the config. I used this here: https://github.com/PrusaMK2Users/MK2_Tips_and_Tricks/wiki/OctoPrint-HAProxy-Config

I've contacted the person who wrote it and he said he should be able to modify it to add your initial suggestion.

closed 2018-01-13 by devsfan1830

MoonshineSG, 2018-01-13:
instead of `octoprint.local`  use  the real  IP or URL that you use to access it from outside network. 

but if the config gets updated as suggested, it's good enough... 

Just remember to use HTTP**S** !