MoonshineSG/OctoPrint-Mobile #64: Login feature on the app?

72428, 2018-02-15:
Hi, 

I bought your app and i like it. Its in depth and cool looking but what does worry me is that I do not get a login option? if someone is able to get into my ocotprint, will they be able to access it? Or is this what the API key is for? 

Kind regards,
Robbert

72428, 2018-02-15:
Is it even possible to add a SSL certificate to your octoprint?

MoonshineSG, 2018-02-15:
Most people would add it to **haproxy** which is extremely easy and tons of "how to" available on the internet (ex: https://printoid.net/access-octoprint-from-the-internet/). Read https://github.com/MoonshineSG/OctoPrint-Mobile/wiki/SSL-Error if you do a self signed cert, but I would suggest a "Let's Encrypt" cert that is free and fairly well accepted. Again, tons of tutorials on how to create and automatically renew (Let's encrypt certs have a short 3 months life span - but can be automatically renewed!) 

P.S. SSL and HTTPS are the same thing (in the given context)

closed 2018-02-20 by MoonshineSG