MoonshineSG/OctoPrint-Mobile #65: [Request] Finish print notification

gamernjs, 2018-02-20:
Hello,

I wasn’t sure where to post this. Would it be possible to have the app notify you when the print is finished? Or is that already a feature that I’ve overlooked? 

Thank you, 
- Noah

MoonshineSG, 2018-02-21:
Absolutely. It's one of the nice features of Nautilus. 
See https://github.com/MoonshineSG/OctoPrint-Mobile/wiki/Notifications for details. 
Post here if you have problems with it and I'll be happy to guide you in solving it....

gamernjs, 2018-02-21:
Awesome! I figured I overlooked it. Thank you! Do you plan to add Apple Watch Support? I'd be happy to beta test it if so.

MoonshineSG, 2018-02-21:
To be honest, I don't have such plans. I don't have one nor plan to get one - although I am a full apple ecosystem :D 
And it would be quite complicated with the multi printer. 
Sorry.

closed 2018-04-10 by MoonshineSG