MoonshineSG/OctoPrint-Mobile #67: Issue with Original Prusa mk2 and Mk3

jesterino, 2018-05-31:
Hello 
i have 6 printers and 4 work fine with Nautilus.
I have issue with mk2 and mk3. After connected an error report "Server Offline attempt to reconnect (9). Retrying it work. If i go on the Camera it show me few frame and after the pics freeze.
It happen with both printers with Octopi 0.13 and 0.15

MoonshineSG, 2018-06-01:
more importantly is the OctoPrint and nautilus plugin versions you're running. 
Is there anything "special" about the config on those 2 printers ? 
Is the URL correct ? protocol, address, port... ? If you connect using a browser on your phone, does it connect (to the full OctoPrint UI ) ?

CADMunkie, 2018-08-05:
I have the same issue I am running the latest Prusa Octopi and the latest OctoPrint from 05-08/2018 

It appears that installing Nautilus prevents the Octoprint from booting on a Prusa i3MK3 

"Looks like something went wrong during startup, the server is gone again. You should check octoprint.log." 

There is no way of accessing the octoprint.log as I cannot get the thing to boot fully to search for files

Hope this helps

James

MoonshineSG, 2018-08-06:
I didn’t have time to upgrade yet, but i’ll Try to do it by next week and see what’s wrong then fix it.

MoonshineSG, 2018-08-10:
@CADMunkie disabling Nautilus allows Octoprin to boot normaly ?

closed 2019-01-20 by MoonshineSG