MoonshineSG/OctoPrint-Mobile #70: REQUEST notification on M600 command - Filament change

AMFkuna, 2018-10-28:
Hi,
Can You implement M600 please?

On colour change command.

Thank You

MoonshineSG, 2018-10-28:
Can you elaborate? Isn’t that just like “pause”? (I don’t use it, so I am not sure how it works)

AMFkuna, 2018-10-29:
Look at this page https://www.prusaprinters.org/color-print/

AMFkuna, 2018-10-29:
I have printers at other room and if I print color object, color change notification will be useful

MoonshineSG, 2018-10-29:
so... what exactly do you need ? notification for M600 ? thats relatively easy to implement...

AMFkuna, 2018-10-29:
Yes, if M600 command is present on gcode, send notification "Please change filament." or user text from configuration.

MoonshineSG, 2018-10-29:
done

MoonshineSG, 2018-10-29:
update and enjoy!

AMFkuna, 2018-10-29:
OK 
Now I'm printing, I'll test it asap.

Thank You very much for fast reply.

MoonshineSG, 2018-11-11:
managed to test it ?

AMFkuna, 2018-11-11:
Hi, it working perfectly
many thanks

closed 2018-11-13 by MoonshineSG