MoonshineSG/OctoPrint-Mobile #71: ERROR: Icloud not avalliable

1Ghasthunter1, 2018-11-22:
Hello,
Whenever I want to add a second configuration, it always returns with a pop-up window saying:
ERRORL iCloud not avaliable.
Purchasing additional configuration slots requires a valid icloud account. Etc.
How do you fix this?

MoonshineSG, 2018-11-23:
As the message says, enable iCloud

1Ghasthunter1, 2018-11-23:
I’m not too sure what that means or how to do it. I am signed in to my iCloud on the device and everything works with iCloud signed in.

MoonshineSG, 2018-11-23:
Hmm... maybe you denied access for the app to access iCloud?

1Ghasthunter1, 2018-11-23:
Okay, problem solved. I had to turn on iCloud Drive. From there I enabled nautilus. Thanks!

1Ghasthunter1, 2018-11-23:
Im using the iphone XR on ios 12.1.1 beta. But enabling that gave me control over the apps

closed 2018-12-03 by MoonshineSG