MoonshineSG/OctoPrint-Mobile #72: Nautilus is stuck at loading screen since last OctoPrint update

Hedrakk, 2018-12-13:
Hi!

Just to mention that since i've updated my OctoPrint to 1.3.10, Nautilus is just stuck at the loading screen.

Thanks

MoonshineSG, 2018-12-13:
i just updated to 1.3.10 to test and the Nautilus plugin doesnt give errors. 

If your Octoprint working well on a normal desktop ? no errors ? Have a look at the logs...

Hedrakk, 2018-12-13:
Hi!

Yeah that’s what is strange: Octoprint does work well when I’m connecting through my desktop, everything is fine. 

I tried to delete and reinstall Nautilus why no more success. 

Jean-Francois Chabot
__________________________________
Capitaine
Equipe Extra-Life des Bionic Marmots|
Bionic Marmots Extra Life Team|
www.extra-life.org/participant/jfchabot

MoonshineSG, 2018-12-13:
Is it over internet or local? Can you use browser on the phone to load octoprint?

Hedrakk, 2018-12-13:
This is all local. And I did try to load Octoprint on my phone with the web browser and it’s working too. 

I’m printing right now so I don’t to mess too much with octoprint’s setting for now. But once my print is done, I’ll try to generate a new Octoprint API key and to reconnect with Nautilus using the new key. 

Jean-Francois Chabot
__________________________________
Capitaine
Equipe Extra-Life des Bionic Marmots|
Bionic Marmots Extra Life Team|
www.extra-life.org/participant/jfchabot

Hedrakk, 2018-12-13:
> Incorrect key would give an error message... “stuck” is strange... timeouts and errors should be handled and reported properly...
> very strange

It’s strange indeed...

You were mentioning that I should take a look at the log file. Can you tell me which one and where can I find it?

Thank

Jean-Francois Chabot
__________________________________
Capitaine
Equipe Extra-Life des Bionic Marmots|
Bionic Marmots Extra Life Team|
www.extra-life.org/participant/jfchabot

MoonshineSG, 2018-12-13:
Logs can be accessed via ssh or via octoprint settings 
Look under octoprint.log for errors (you would recognise one if you see one)

MoonshineSG, 2018-12-13:
honestly, i can't say for sure. i don't know what's wrong and why do you get the error ...

Hedrakk, 2018-12-13:
I just realized... I’m on iOS beta program and I upgraded to iOS 12.1.2 a few days ago... that might be related. 

Jean-Francois Chabot
__________________________________
Capitaine
Equipe Extra-Life des Bionic Marmots|
Bionic Marmots Extra Life Team|
www.extra-life.org/participant/jfchabot

Hedrakk, 2018-12-14:
I’ve reinstalled everything without more success. One strange thing though is that I’m still getting my end of printing notifications. 

I don’t know if it is possible for you to test Nautilus on iOS 12.1.2?

Thanks

Jean-Francois Chabot
__________________________________
Capitaine
Equipe Extra-Life des Bionic Marmots|
Bionic Marmots Extra Life Team|
www.extra-life.org/participant/jfchabot

Hedrakk, 2018-12-14:
When I tried to uninstall/re-install Nautilus Plugin on my Octoprint, I noticed that all my Nautilus command customizations were back. 

So I guess some Nautilus files remained on the Octopi even after the uninstall. Is there a folder or bunch of files that I would need to manually delete to get rid of any traces of Nautilus before re-installing it. 

Thanks

Jean-Francois Chabot
__________________________________
Capitaine
Equipe Extra-Life des Bionic Marmots|
Bionic Marmots Extra Life Team|
www.extra-life.org/participant/jfchabot

MoonshineSG, 2018-12-14:
I’m not sure how the uninstall works, but you can remove/rename the folder .octoprint/data/nautilus (octoprint mobile) I don’t have access to a computer and it’s been a while since I touch the codes)

Unfortunately I am at the airport leaving on a holiday so it’s gona be tough to help - but i’ll Try if it’s urgent

Hedrakk, 2018-12-14:
There’s no life threatening situation here, don’t worry.

It’s just that using my printer without the Octoprint/Nautilus combo is like eating a peanut butter-jelly sandwich without the jelly; there’s something missing 😊 

Thanks

Jean-Francois Chabot
__________________________________
Capitaine
Equipe Extra-Life des Bionic Marmots|
Bionic Marmots Extra Life Team|
www.extra-life.org/participant/jfchabot

uhu2001de, 2018-12-15:
The problem is the login with the new version  OctoPrint 1.3.10. Your app has no login.

Frank

MoonshineSG, 2018-12-15:
No login needed. Authentication is done via API Key

keithellis74, 2018-12-22:
Yep, I have exactly the same issue, help appreciated, I’m on OctoPi 1.3.10 and iOS 12.1

MoonshineSG, 2018-12-22:
Sorry guys, on holiday... early jan i’ll have a look and fix it...

sajovicd, 2018-12-26:
Have the samae problem, just since 1.3.9 version of octopi.
Hope you can fix it when arriving home safely.

MoonshineSG, 2019-01-02:
1.3.10 has introduced "application keys". until i managed to support that feature make sure API key support is still enabled

MoonshineSG, 2019-01-10:
i can now replicate the problem and disabling the plugin "Force Login" allows Nautilus to work.

 I'll work over the next few days to come with a permanent solution, but meanwhile, disable the plugin.

sajovicd, 2019-01-10:
Thanks, it is working now with that plugin disabled.

MoonshineSG, 2019-01-19:
plugin and ios app will both require an update (just make sure your plugin is at 1.24, not 1.23 -  i made a mistake in the commit)

These two contain a workaround fix and a final fix that will work with the future 1.3.11

Report back if there are problems...

MoonshineSG, 2019-01-20:
Make sure you have iOS app version 2.1.7 and plugin version 1.24 and this issue is solved.

closed 2019-01-20 by MoonshineSG

MoonshineSG, 2019-01-20:
Thanks everyone for you support and patience.

PRGeno, 2019-01-21:
Works perfectly again.  Thanks!

sajovicd, 2019-01-21:
Perfect.