MoonshineSG/OctoPrint-Mobile #73: code 401 in iOS app

DBa2016, 2018-12-13:
When trying to add the printer to Nautilus iOS app (and after passing the hurdle of trusting the certificate), I get a message stating "The configured URL does not point to a running instance of OctoPrint with Nautilus plugin installed. [code 401]".
I have OctoPrint running behind a haproxy with basic auth enabled - if that matters.

Both web browsers and "OctoClient" app have no trouble connecting to my printer.

MoonshineSG, 2018-12-13:
http authorisation is not currently supported
You need to add a haproxy rule to bypass authentication for user agent starting with “nautilus”

As for https certificate, I suggest you look into “let’s encrypt” free and fully supported certificates

DBa2016, 2018-12-13:
Well, this makes the app entirely unusable - obviously, "user agent" is an easily modifiable field (you can even modify it with a simple browser plugin), so by letting haproxy bypass the authentication you are basically letting anyone in who has as little criminal intent as installing a browser plugin. 

By when do you plan to support http auth?

MoonshineSG, 2019-01-02:
the API key is the "authentication". it is more safe than http auth. of course "https" is a must if you open acccess to the wild internet

closed 2019-01-20 by MoonshineSG