MoonshineSG/OctoPrint-Mobile #75: feature requests

treflip, 2019-04-21:
Where else to post feature requests?

I hate landscape mode for my 7+. It's hard to hold the phone with one hand and select things this way. I almost drop my phone all the time (do you know how much these screen costs lol!)

I read you force this option to show all the stuff. I understand that but I have not bought this app yet. It may be easier than the web app but I am not sure I will like it now that I found that out.

Will you consider making a portrait mode option? I would very much like that better than landscape!

Thanks!

MoonshineSG, 2019-04-22:
sorry, unlikely that portrait will be made available...

closed 2019-04-22 by MoonshineSG

treflip, 2019-04-22:
Understood. Thank you