MoonshineSG/OctoPrint-Mobile #76: Code 403 issue

OutspokenPlanet, 2019-05-20:
Recently updated the stretch light installs on my four separate pi servers, and now receiving the message
"The configured URL does not point to a running instance of OctoPrint with the plugin "Nautilus" installed. [code 403]"

I've restarted all of my network infrastructure, and bypassed my haproxy server, all four instances return the same error.

OctoPrint 1.3.11 running on OctoPi 0.16.0 - on one machine
OctoPrint 1.3.11 running on OctoPi 0.15.1 - on three machines
Nautilus (1.24) - all machines

Is there any further info I can give you to help?

keithellis74, 2019-05-20:
Yep, I’m getting the same message after updating OctoPrint to 1.13.11

Keith Ellis

MoonshineSG, 2019-05-20:
I will upgrade, investigate and get back to you with a fix in a couple of days

MoonshineSG, 2019-05-20:
OK... I updated to 1.3.11 and using the latest Nautilus (v 1.24) and there's no problem with it... 

In this case I will need some logs from your side to see if there are any error messages and some more info regarding your setup. 
Are you using anonimous login or you have a user created? 
A list of plugins to see if there could be any conflicts. 
If you go to the Octoprint Settings page do you see Nautilus preferences page ? (meaning the plugin is loaded) 

waiting for your info

keithellis74, 2019-05-20:
Mine worked initially, try rebooting.  I’ll get the info requested to you when I get a chance. Thank you.

Keith Ellis

MoonshineSG, 2019-05-20:
seams to be related to ForceLogin plugin...(if yiou disbale that, it's all good!) 
 I'll contact the developer (of OctoPrint) and work out a solution...

MoonshineSG, 2019-05-21:
a fix is with Apple for review. Should be out as soon as is approved.

MoonshineSG, 2019-05-21:
fix coming soon. pending apple approval.

nevergitithub, 2019-05-21:
Hi. I am an absolute beginner. After installing the latest version of Nautilus and Octoprint a couple of days ago, I’m getting the same message. No mods or other plugins are installed than Nautilus. I wil check the ForceLogin this evening.

MoonshineSG, 2019-05-21:
It’s a bug in octoprint but a workaround fix has been submitted to apple... give it a couple of day , max...

MoonshineSG, 2019-05-21:
The update is out. Upgrade to the latest - v2.1.8 - and the problem will be gone. Thanks.

closed 2019-05-21 by MoonshineSG

keithellis74, 2019-05-21:
Seems to be fixed, thank you.

Keith Ellis

OutspokenPlanet, 2019-05-25:
Awesome job, thank you

nevergitithub, 2019-05-25:
Works fine now. Thank‘s a lot.