MoonshineSG/OctoPrint-Mobile #78: Nautilus app currently not available in your coutry

sajovicd, 2020-09-04:
Hello,

Just try to install Nautilus app on Iphone 11 and I always got error "App is not available in your country." 
I live in Slovenia / Europe. But I change to US store and then I got error "app is not available in US store"

Any help?

MoonshineSG, 2020-09-05:
i removed it from appstore as i had no time to keep it updated and many other apps are now available. when i made this there was nothing that allowed a decent octoprint UI on a smart phone 
I'll probably open up the source , but you'll need a developer account to install it on your phone

keithellis74, 2020-09-08:
That’s a shame. I really like this app. I’ll have to make sure I have it backed up.
Thanks for all your efforts in this app.
Good luck.

Keith Ellis

MoonshineSG, 2020-09-08:
many chnages on the apple side and very weak sales :D 
I'm glad ot hear you like it, but sadly, no time to keep it up to date...

closed 2020-09-08 by MoonshineSG